grbl/grbl #1684: Wiki "Gcode Examples" is empty

max-hans, 2020-11-19:
Somehow the content of "GCode-Examples" must have gotten lost. Currently there is only a welcome message in Chinese (欢迎)
[G-Code Examples](https://github.com/grbl/grbl/wiki/G-Code-Examples)

Previous revisions seem to have contained some information.
For example → [5f9978fca6d1c416eff918e0c19ebf2a682a41d7](https://github.com/grbl/grbl/wiki/G-Code-Examples/5f9978fca6d1c416eff918e0c19ebf2a682a41d7)

MeJasonT, 2020-11-19:
have some of mine

[Milling -CNC Projects.zip](https://github.com/grbl/grbl/files/5566912/Milling.-CNC.Projects.zip)

Check the dimentions, Z heights and set  work offsets before trying them out - you will crash otherwise.

Try and get a program called camotics
you will be able to simulate the parts before atempting to run any jobs
[https://camotics.org/](url)

also zero your z up in the air (cut in air) before setting it down on the material surface and setting Z heights.
this is often used by pro machinists to prove a tool path before commiting to milling a job - you should b able to see if the code will run as expected and wont get in the way of any clamps etc.  a couple of the examples have been cut out of flat steel plate with the plate (with enough extra stock) clamped in the vice.

all NC was produced using CamBam and a GRBL postprocessor
[http://www.cambam.co.uk/](url)

One finalm note this forum is no longer supported
the new grbl site is [https://github.com/gnea/grbl/issues](url)

there are only a few grumpy old men on here who refuse to move

max-hans, 2020-11-25:
thanks very much, @MeJasonT ! I personally don't really need sample files. I just wanted to give feedback =) however, this is probably helpful for anybody who needs some stuff to get started =)