grbl/grbl #1685: no speed control while in imaging mode

BTUTTLE1943, 2020-11-22:


MeJasonT, 2020-11-22:
try asking your question here
this site isnt being supported much anymore
[https://github.com/gnea/grbl/issues](url)

MeJasonT, 2020-11-22:
not quite sure what you mean by Imaging mode

BTUTTLE1943, 2020-11-23:
you can run in line mode or image mode for photos you use image mode. for what ever reason this laser slows down the pulse a   with when it goes to heaver burn

terjeio, 2020-11-23:
> not quite sure what you mean by Imaging mode

My guess is that the issue is about the programmed feed rate not beeing reached when rendering an image with many short moves. If so then this is mainly due to a small input buffer (or not depending on how the sender operates) and not enough entries in the planner buffer.

I did a short test  recently with a 32-bit version with plenty of RAM where I increased number of entries to 300 (from the default 16) leading to a significantly higher feed rate. An even higher feed rate was achieved when using count mode transfer with a larger input buffer.

Further details [here](https://embeddedtronicsblog.wordpress.com/2018/12/15/stm32-grbl/) , [here](https://embeddedtronicsblog.wordpress.com/2019/08/03/testing-mini-gerbil-firmware/) and [here](https://embeddedtronicsblog.wordpress.com/2020/07/20/grblhal/).