grbl/grbl #1686: M3 is the only command that shuts the laser off, does not shut off during travel movements, GRBL 1.1f, does not shut off when print is finished

titus123abc123, 2020-11-23:
Hi all

I've hit a bit of a roadblock with my build. M4 S### turns on my laser, as I would expect it to. M4S0 does not turn it off, nor does any other command I've tried. M3 followed by any S value is the only command that will shut the laser off. The problem with this is that all my prints have lines through them as the laser moves and doesn't shut off. After it finishes a task it just sits there with the laser on and burns a hole. Adding M3 to my end gcode does make it turn off at the end of a task. Any ideas of what to do?

I have the TTL pin of [this laser](https://www.aliexpress.com/item/4001027841928.html?spm=a2g0s.9042311.0.0.27424c4dNojQBS) wired to the spindle direction pin of the shield from [this kit](http://www.zyltech.com/arduino-cnc-kit-w-uno-shield-stepper-motors-drv8825-endstop-grbl/). Am I using the wrong pin? I can't find any useful documentation about the shield.

Laser mode is enabled on the firmware. It shows $32=1 (see  bottom of post for $$ output)

Running GRBL 1.1f (flashed with LaserGRBL if that matters) and Lightburn for gcode (the same thing happens with any host though). 

Let me know if there's any more info you need
Thanks in advance!

$0=10
$1=100
$2=0
$3=2
$4=0
$5=1
$6=0
$10=10
$11=0.020
$12=0.010
$13=0
$20=0
$21=0
$22=0
$23=7
$24=50.000
$25=300.000
$26=250
$27=5.000
$30=255
$31=0
$32=1
$100=160.000
$101=160.000
$102=0.001
$110=4500.000
$111=4500.000
$112=500.000
$120=90.000
$121=90.000
$122=10.000
$130=320.000
$131=320.000
$132=150.000

zanglerecke, 2020-12-22:
possible, that the signal is inverted:
m3 should switch the laser ON
m4 waits for movement. while move it will be switched on
m5 deactivates laser

try to invert the signal (e.g. a fet with an pullup)

MeJasonT, 2020-12-22:
At the beginning of your code set S*** to the max you use to turn the laser on 1000?
after its first use it remains a constant value in memory so you dont have to send the S**** command again
M3 should switch it on if you are wired as spindle fwd  M5 will switch it off
your postprocessor(CAM) should handle the laser on/off commands whilst traveling
consider a postprocessor written for a plasma cutter if you cant find one for laser for you machine

MeJasonT, 2020-12-22:
ok the word i was looking for is Modal - when a command is persistant.  ie it stays constant until its changed in value
spindle speed, feedrate and G* moves are all modal
so if you send G1 then you dont need to tell the machine on every line that is is making G1 moves, when you need to rapid G0 then you send G0 and it replaces the G1 modal and becomes the persistant command.
If S1000 is sent to the machine controller then the spindle speed is modal so turning it on and off with M3(or M4) and off with M5 you dont need to send a spindle speed command again (unless you want to change speed obviously)
Normal machine practice would be to sent Feed, speed and direction command on a single line of code F200 S1000 M3, you would mill the section of the part then M5 to stop the spindle ie direction zero/off  to start milling again M3 The speed and feed are still set in modal memory.  There are a couple of other modal functions, home position, G54-G58 work coordinates, tool height and tool change location and lets not forget machine coordinates which you will never mess with as it is the refference of all other coordinates on your machine.

MeJasonT, 2020-12-22:
what CAM program are you using and what postprocessor is it using
CAM creates your tool paths from a list of machine operations MOPs these paths are turned into gcode by the postprocessor.  all machines and manufacturers use different command structures and have different bed sizes and mechanics so a postprocessor is written to be the interface between CAM and your machine.  It will look for path instructions like rapid to start of a path/job The PostProc. knows what instruction your machine needs to make that move possible on your controller - in our case GRBL but could be Haas, Siemens, Fanuc etc  the latter using a letter and line numbers for each line of code, Grbl hates line numbrs and gets very annoyed if you use them.  in your CAM package the postprocessor will decide when to turn on your laser with M3 and off with M5 whenever its required to do so on the tool path within a MOP.  In milling you might face your material, make a pocket and then machine the outside profile - these are classed as machine operations, you could have sevral paths within a single operation ie engraving letters into your part.  Between each path you will likely have a rapid move to the next start point.  This is when you expect the laser to be turned off -  Postprocessor, thats the term you need to understand and learn to get your head around how to turn on and off your laser when traveling.