grbl/grbl #1693: Y and Z axis work at the same time,even i send command only for 1 axis

allyerkan, 2020-12-18:
Hi all,
i have a problem in y and z axis. ı am using arduino uno+cnc shild with GRBL compiled. .If send command to move only the y axis in both(+/-) directions, z axis goes + direction too at the same time with y axis. and if i try to move z axis only in both directions this time y axis moves only + direction, again at the same time with z axis. btw I used 2 step motor for y axis 

There are several threads i read from forums but i didnt have a solution for this problem
Is there anyone having problems similiar to mine?

PicEngraver, 2020-12-20:
Are you sure of your wiring connections?  What Arduino (clone?)/shield/controller are you using?  What grbl version?  
No one can possibly guess what you are doing, or what you are doing wrong.  Include more specific details of your system and setup.

PicEngraver, 2020-12-22:
I do not have the same setup as you list, but perhaps site this will help:
https://www.electroniclinic.com/arduino-cnc-shield-v3-0-and-a4988-hybrid-stepper-motor-driver-joystick/

allyerkan, 2020-12-23:
thanks for the reply @PicEngraver 

I read nearly all articles in this site and there is nothing i can find about my problem
I did all the steps according to the guides but i dont understand why y and z axis move simultaneously, even i send the code for 1 axis. In an article, someone mentioned the resoldering arduino pins but i am not capable of that. I will buy  new ones, i think

PicEngraver, 2020-12-23:
You may have a corrupted version of grbl.  Maybe try this version???
https://github.com/gnea/grbl/releases

allyerkan, 2020-12-23:
i used that one aldready but result is same.

PicEngraver, 2020-12-23:
Grbl 1.1h, released August 2019 is good.  

So, assuming you have made no changes to the grbl source code, this would then appear to leave either a bad wiring connection or a bad shield.
  
This is a very good driver shield:
https://synthetos.myshopify.com/products/gshield-v5

And I have one of these running my small test machine:
https://www.amazon.com/gp/product/B08K34T4Q2/ref=ppx_yo_dt_b_asin_title_o00_s01?ie=UTF8&psc=1
Just be aware that the Z axis limit switch input and the PWM outout pins (11/12) are the old setup and need to be swapped when you wire up the limit switches.

allyerkan, 2020-12-23:
thanks again @PicEngraver 

yes i didnt make any changes in grbl. wiring is ok also.

I am thinking to change the shield too as you suggested, that's the only solution come to my mind.

PicEngraver, 2020-12-23:
Wishing you the best.  Merry Christmas.

allyerkan, 2020-12-25:
thanks for kind help @PicEngraver . Mery Christmas

PicEngraver, 2020-12-25:
And to you and yours as well.