grbl/grbl #1695: Is grbl resetting itself? Terminal prompt when sending gcode

balthisar, 2021-01-03:
When trying to send commands, such as moving my Y axis, I'll sometimes get this in the terminal output:

    Grbl 1.1h ['$' for help]

Usually when that happens, no further gcode is processed. I don't have to reboot the Arduino, but unplugging and replugging the USB seems to get things in working order again.

Is this a sign of bad flash? Or maybe the USB connection is dropping? Or the Arduino needs a firmware update per the troubleshooting page? 

Associated with this, the coordinates are reported with really, really big numbers. Not MAXINT as far as I can tell. It usually happens when I try to jog more than a few millimeters. In any case, I'll do additional troubleshooting, so my question really is that above: what's it mean when the terminal spontaneously outputs that message?

langwadt, 2021-01-03:
what are your settings?

balthisar, 2021-01-03:
My settings:

~~~
$0 | 10 | step pulse, usec
$1 | 255 | step idle delay, msec
$2 | 0 | step port invert mask:00000000
$3 | 0 | dir port invert mask:00000000
$4 | 0 | step enable invert, bool
$5 | 0 | limit pins invert, bool
$6 | 0 | probe pin invert, bool
$10 | 3 | status report mask:00000011
$11 | 0.02 | junction deviation, mm
$12 | 0 | arc tolerance, mm
$13 | 0 | report inches, bool
$20 | 0 | soft limits, bool
$21 | 0 | hard limits, bool
$22 | 0 | homing cycle, bool
$23 | 0 | homing dir invert mask:00000000
$24 | 180 | homing feed, mm/min
$25 | 1800 | homing seek, mm/min
$26 | 250 | homing debounce, msec
$27 | 5 | homing pull-off, mm
$100 | 2000 | x, step/mm
$101 | 2000 | y, step/mm
$102 | 2000 | z, step/mm
$110 | 2000 | x max rate, mm/min
$111 | 2000 | y max rate, mm/min
$112 | 2000 | z max rate, mm/min
$120 | 150 | x accel, mm/sec
$121 | 150 | y accel, mm/sec
$122 | 150 | z accel, mm/sec
~~~

I'd be surprised if this were noise over USB; I'm using a 1 foot USB A to Mini B connector from the Pi 4 to the Arduino. I've ordered a 450 mm version of the same cable with ferrite cores just to be safe, but I don't know if this is a noise problem or not based on the evidence, which is why I'm asking. In what conditions would GRBL output the message, indicated, to the terminal?

quotelawrence, 2021-01-03:
could someone merely go into grbl v .9 and enter commends 
$130=255
$131=5
$132=0
would this be enough to start the the laser function or does this have to be a higher version grbl

langwadt, 2021-01-03:
2000step/mm* 2000mm/min / 60sec = 66kHz that is more than double the roughly 30kHz GRBL is capable of