grbl/grbl #1696: X and Y not homing

alarebel, 2021-01-06:
I recently assembled, flashed, and powered up my cnc using grbl-1.1h.20190825. I verified that the limit switches were working by pressing them while the machine was not moving, and all 3 axes give the hard limit alarm. When homing, the Z axis stops on the limit switch, backs off and moves in for the second touch off. However, the X and Y axes do not stop on the limit switches. They keep driving the stepper motors even though the switches are depressed. 

 I set $23 to 3 because I need X and Y to home in the negative direction and the Z to home in the positive direction. The limit switches are wired NO and C. The limit switches are on X-, Y-, and Z+. Pins 11 and 12 have been swapped for Z axis homing. Any help is much appreciated.

$0 | 10 | microseconds | Step   pulse time
$1 | 25 | milliseconds | Step   idle delay
$2 | 0 | mask | Step   pulse invert
$3 | 0 | mask | Step   direction invert
$4 | 0 | boolean | Invert   step enable pin
$5 | 0 | boolean | Invert   limit pins
$6 | 0 | boolean | Invert   probe pin
$10 | 1 | mask | Status   report options
$11 | 0.01 | millimeters | Junction   deviation
$12 | 0 | millimeters | Arc   tolerance
$13 | 0 | boolean | Report   in inches
$20 | 1 | boolean | Soft   limits enable
$21 | 1 | boolean | Hard   limits enable
$22 | 1 | boolean | Homing   cycle enable
$23 | 3 | mask | Homing   direction invert
$24 | 25 | mm/min | Homing   locate feed rate
$25 | 500 | mm/min | Homing   search seek rate
$26 | 250 | milliseconds | Homing   switch debounce delay
$27 | 1 | millimeters | Homing   switch pull-off distance
$30 | 1000 | RPM | Maximum   spindle speed
$31 | 0 | RPM | Minimum   spindle speed
$32 | 0 | boolean | Laser-mode   enable
$100 | 400 | step/mm | X-axis   travel resolution
$101 | 400 | step/mm | Y-axis   travel resolution
$102 | 400 | step/mm | Z-axis   travel resolution
$110 | 500 | mm/min | X-axis   maximum rate
$111 | 500 | mm/min | Y-axis   maximum rate
$112 | 500 | mm/min | Z-axis   maximum rate
$120 | 10 | mm/sec^2 | X-axis   acceleration
$121 | 10 | mm/sec^2 | Y-axis   acceleration
$122 | 10 | mm/sec^2 | Z-axis   acceleration
$130 | 200 | millimeters | X-axis   maximum travel
$131 | 200 | millimeters | Y-axis   maximum travel
$132 | 200 | millimeters | Z-axis   maximum travel

alarebel, 2021-01-07:
Limit switch wasn't hitting the arm... it works correctly haha.

closed 2021-01-07 by alarebel

plceleste, 2024-01-20:
Hi, I have a problem with my homing using ugs. Same settings with you I have my limit switches positioned in X-, Y+ and Z+ but whenever I click the home button only the z-axis moves downward and stops (it doesn't perform homing) and an error message pop-up (An error was detected while sending '$H': ALARM:8,Homing fail. Streaming has been paused.
Error while processing response <ok>: An unexpected command was completed by the controller.
Grbl 1.1h ['$' for help]
[MSG:'$H'|'$X' to unlock]) can you help me?

plceleste, 2024-01-20:
here's my firmware settings 

$0 | 10 | Step pulse time
-- | -- | --
$1 | 25 | Step idle delay
$2 | 0 | Step pulse invert
$3 | 6 | Step direction invert
$4 | 0 | Invert step enable pin
$5 | 0 | Invert limit pins
$6 | 0 | Invert probe pin
$10 | 2 | Status report options
$11 | 0.010 | Junction deviation
$12 | 0.002 | Arc tolerance
$13 | 0 | Report in inches
$20 | 0 | Soft limits enable
$21 | 0 | Hard limits enable
$22 | 1 | Homing cycle enable
$23 | 1 | Homing direction invert
$24 | 25.000 | Homing locate feed rate
$25 | 500.000 | Homing search seek rate
$26 | 250 | Homing switch debounce delay
$27 | 1.500 | Homing switch pull-off distance
$30 | 10000 | Maximum spindle speed
$31 | 0 | Minimum spindle speed
$32 | 0 | Laser-mode enable
$43 | 0 |  
$44 | 0 |  
$45 | 5 |  
$46 | 0 |  
$51 | 8192 |  
$52 | 1000066 |  
$53 | 578618160 |  
$54 | 1561217489 |  
$55 | 1000380 |  
$56 | 4224403738 |  
$57 | 107132084 |  
$58 | 4607050 |  
$59 | 0 |  
$60 | 0 |  
$61 | 0 |  
$100 | 800.000 | X-axis travel resolution
$101 | 800.000 | Y-axis travel resolution
$102 | 800.000 | Z-axis travel resolution
$103 | 50000.000 |  
$104 | 50000.000 |  
$105 | 3200.000 |  
$110 | 2000.000 | X-axis maximum rate
$111 | 2000.000 | Y-axis maximum rate
$112 | 500.000 | Z-axis maximum rate
$113 | 500.000 |  
$114 | 500.000 |  
$115 | 500.000 |  
$120 | 3000.000 | X-axis acceleration
$121 | 3000.000 | Y-axis acceleration
$122 | 5000.000 | Z-axis acceleration
$123 | 10.000 |  
$124 | 10.000 |  
$125 | 10.000 |  
$130 | 270.000 | X-axis maximum travel
$131 | 170.000 | Y-axis maximum travel
$132 | 15.000 | Z-axis maximum travel
$133 | 200.000 |  
$134 | 200.000 |

breiler, 2024-01-20:
These settings tells me that you are not using the "standard" GRBL. Without knowing exactly which GRBL you are using I have to make assumptions.

The settings $51-$58 (whatever they are, since they aren't standard GRBL either) indicates that your EEPROM needs to be reset through `$RST=*`.

The reason you are getting the alarm when homing is because GRBL could not find the limit switch within the max search distance. This is determined by the settings $130-$132. For Z this is set to 15 mm.

plceleste, 2024-01-20:
What do you mean by the GRBL could not find the limit switch within the max search distance?

plceleste, 2024-01-20:
My hard limit/physical switch is working fine whenever I manually check it, it alarms (as it should be).

plceleste, 2024-01-20:
And it is normal to get an alarm whenever I connect my cnc to the ugs? (I already connect my limit switches the connect my cnc to the ugs software)
[image]

plceleste, 2024-01-20:
[image]
[image]
 as you can see, my reference/origin/home point is at the upper lower part of the platform. I put my limit switches on, X- (left), Y+(front) and Z+(up).

plceleste, 2024-01-20:
[image]
here is my setting for limit switches

plceleste, 2024-01-20:
[image]
as well as my soft limits

plceleste, 2024-01-20:
and also I can't move/jog manually the cnc to X+, Y+ and Z+. I think there is something else with this 
[image]
the numbers below the real location (X = 0.000, Y= 0.000 and Z= -0.500)

plceleste, 2024-01-20:
I am new to this CNC 3018 pro machine. I hope you can help me

plceleste, 2024-01-20:
whenever I  click the home it is the result 
[image]
only the Z-axis moves downward for 1mm I think

breiler, 2024-01-20:
It sounds like You are trying with too many features at once, start simple and add capabilities once you have got one feature working. Start over by disabling homing, soft limits and make sure that you can jog the machine. Then enable homing and get that working. Then enable soft limits.

> What do you mean by the GRBL could not find the limit switch within the max search distance?

This setting has two usages, it will define your machine work area and provide soft limits (if it is enabled). It will also be used to let the homing feature know how far it should search for the limit switch. If it is further away (in this case 15 mm) it will throw an error saying it could not find the limit switch. Increase that distance.
[image]

> And it is normal to get an alarm whenever I connect my cnc to the ugs?

Yes, this is the default behaviour of GRBL. If you have homing enabled and this setting https://github.com/grbl/grbl/blob/master/grbl/config.h#L58 it will enter an alarm state  when booting the controller. It is basically warning that it is at a unknown position and needs to home before it can be safely used.

> whenever I click the home button only the z-axis moves downward and stops

Since you have configured your homing to search in Z+ direction indicates that your Z axis is inverted. Make sure that you can jog the machine in the correct directions.

plceleste, 2024-01-23:
[image]
when I try to trigger only one of the limit switches, all the axes become red here in the limit switch settings. why is that?

plceleste, 2024-01-23:
I found out that only the y-axis limit switch is recognizable maybe because only the y-axis has a noise filter capacitor on the board. I will also try to put a noise filter capacitor on the x and z-axis. Hope it will work

plceleste, 2024-01-23:
I am done putting the noise filter capacitor on the board for both the x and z-axis but whenever I manually trigger 1 limit switch on the setting all of the switches still turned red

breiler, 2024-01-23:
> when I try to trigger only one of the limit switches, all the axes become red here in the limit switch settings. why is that?

Without knowing what board you have, which GRBL flavor you are using, or how you have wired things I can't answer this. Best guess is to recheck your wiring and your `config.h`.

Did you reset your EEPROM settings like I asked you? Some settings looked corrupt and I have seen all types of weird behavior when this happens.

plceleste, 2024-01-23:
I am using benbox controller board and grbl1.1. I wired my limit switch to NC.
[image]