grbl/grbl #1702: i cant get the end stop to work  can somebody give me a reply thank you

21091958, 2021-02-03:


1Metalguru, 2021-02-03:
"it doesn't work"  is not enough information to even try to troubleshoot the problem.

21091958, 2021-02-03:
hi thank you for the reply
ok i wire the way it is supose green and black the red is not plog in  i put the $22=1 and the $21=1 and noting start homing and crush the end stop  i can get the suich to work  i look on youtube and i can get the info i need  so your my last hope 
thank you  jean pierre

TIDEngine, 2021-02-15:
Same issue, with me. Only thing you can do is use soft limits for homing and hard limits for protection. In some way when you start homing switches do not respond anymore.

21091958, 2021-02-15:
hi thank you for your reply
so by the look of it we can help one a other i will let you know
what i find somebody tell me it is software whit window 10
i will test that way  i will get back to you

thank you jean pierre

21091958, 2021-03-04:
hi friend

i have the way to conect the Z axes  you have to conect pin Z+ to pin 12 on the clone axes side of the arduino  let me know if you got your cnc working
thank you jean