grbl/grbl #1715: Trouble with GRBL controlling speed of the spindle through VFD

bretonesf, 2021-03-23:
Hello everyone, I'm building a CNC and while I'm waiting for the mechanical parts, I already built the electrical board which include an Arduino with GRBL v1.1h, DM566 as stepper drivers and one chinese YL-620 VFD to control a 2.2kW water cooled spindle.

I'm doing some test to check my electrical installation with bCNC and here's my problem :

If I set the VFD for control via its display, the spindle is correctly controlled.

If I try to control the speed via the D11 pin of the Arduino (directly from Arduino because the VFD can receive the PWM 5V, or with a 5V PWM to 10VDC converter, tried both ways, same result) and the correct parameters in the VFD, by sending a M3 Sxxx, the targeted frequency is displayed on the VFD (blinking value that matches with the desired speed) so the command is sent properly from the Arduino. 

The problem occurs when I try to close the start command on the VFD (should be a simple button) : the red LED RUN" is ON display stops blinking, proof that it tries to reach the set frequency, but that frequency instantaneously drops to 0 (and the spindle of course doesn't rotate) and the parser show M5 S0.

Do you guys have any idea what's going on? It's now been 2 days I'm looking over the internet to find a solution without any success.

jahnj0584, 2021-03-23:
Check your acceleration settings. Too high and it will alarm out before It
can get to speed. Make the spindle work perfectly first then make it
controlled by arduino.

bretonesf, 2021-03-23:
Thank you for fast answer.

The spindle works without any problem if I don't use the PWM, but only manually on the display of the VFD itself.

When I try with PWM, the voltage from D11 is lost instantaneously as soon as I close the run contact of the VFD, but I have no alarm.

Maybe I misunderstood what you're trying to explain?

jahnj0584, 2021-03-23:
I assume you are ESL, so does "close the run contact" you mean sending the
command?

Does the spindle start, run, and shut off properly when you are ONLY using
the VFD to enable and disable the spindle?

bretonesf, 2021-03-24:
When I use only VFD to control the spindle (param P00.01 at 0), yes it works as mentionned.

About the first question : On the VFD, I have to connect the 0-10V (with converter) from Arduino to VI1 and GND, and then close the contact between XGND and FWD (I just put a wire for the trials) to have the VFD in run mode. 

If I send for example an M3 S12000 from bCNC, with contact FWD open, I can see the targeted frequency of 200Hz blinking on the VFD -> the command is well sent to the VFD. As soon as I close the FWD contact, the VFD is in run mode (red led "RUN" is ON), the display doesn't blink anymore and the frequency drops instantaneously to 0 because the parser switches to M5 S0 by itself -> the spindle of course doesn't rotate.

If I start with contact FWD closed, the VFD is in run mode (red led "RUN" is ON), the display doesn't blink and display is 0Hz which is normal.  If I send for example an M3 S12000 from bCNC, I can see the display varying for half a second, then back to 0, the spindle doesn't rotate, the parser switches to M5 S0.