grbl/grbl #1718: Homing Maschine bleibt stehen.

funboxbolzer, 2021-04-13:
Mir wird folgender Fehler angezeigt wenn ich Home drücke.
Ich habe keine Z-Achse installiert, muss nur X und Y homen.

##
ALARM:9
Referenzfahrt fehlgeschlagen. Endschalter in Suchentfernung konnte nicht gefunden werden. Definiert als 1,5 * max_travel bei der Suche und 5 * Pulloff bei Suchphasen.
##

wie kann ich diesen Fehler verhindern? 
Meine Größe der Arbeitsfläche ist 700x500mm

Danke

Adrianf1980, 2021-04-13:
Would be nice it you would write in English.

Rob2029, 2021-04-14:
Sie können die Anschlüsse der Endschalter überprüfen und separat testen, ohne ein hommig zu machen.
Sie können diese Seite https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#22---homing-cycle-boolean überprüfen und die Einstellungen $ 20, $ 21, $ 22, $ 23, $ 24, $ 25, $ 26, $ 27 überprüfen
Entschuldigung für mein Deutsch, ich verwende Google Übersetzer

funboxbolzer, 2021-04-14:
> Would be nice it you would write in English.
> […](#)

Sorry, I will try and hope you can understand what my problem is. 

I will use homeing. When $21=1 in my config.txt

when i start homing my machine is miving about 5cm and then an error occured. Alarm9. That means something like the way to find the end ist too long (my own translation). 

I have only an X and Y axis no Z. Is there an other config to use probably?

funboxbolzer, 2021-04-14:
> Sie können die Anschlüsse der Endschalter überprüfen und separat testen, ohne ein hommig zu machen.
> Sie können diese Seite [...]

Thank you for reply. Here my Config
`$$
$0=40
$1=40
$2=0
$3=40
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=30000.000
$25=20000.000
$26=250
$27=1.000
$30=1000
$31=0
$32=1
$100=160.000
$101=160.000
$102=160.000
$110=60000.000
$111=60000.000
$112=60000.000
$120=100.000
$121=100.000
$122=100.000
$130=700.000
$131=700.000
$132=200.000`

I have no Z Axis. Is this the probably an problem?

funboxbolzer, 2021-04-14:
Problem is solved.
I changed my $25 from 20000 to 1800.
It was to fast. So now i can home my machine.
Thank you all for your help.

Klaus

Adrianf1980, 2021-04-14:
I think that your set up of your machine is not correct.
I send you a link where it’s described how to make all the calculation that GRBL required to run smoothly.

https://diymachining.com/grbl-settings-101-a-how-to-guide/

Also this is ok to look at:

https://m.all3dp.com/2/gbrl-settings-gbrl-configuration/

Once that your machine is set correctly we can have a look at the homing problem.