grbl/grbl #1721: How to homing only one axis， such as X axis？

wsjlmq, 2021-05-17:
My English is not quit good, please understand.
Due to the needs of the actual project, I only connected a stepper motor on grb.I need to set the hardware limit for the x-axis.But when I used the command "$H" to reset grbl, I found that only three axes could be reset at the same time.
Hope to get your advice,Thank you very much.

wsjlmq, 2021-05-17:
I tried to make changes in the  **_limit.c_** file, but it didn't work

langwadt, 2021-05-17:
config.h line 75-76

wsjlmq, 2021-05-18:
> config.h line 75-76

I think this is the place you said
**_// Enables single axis homing commands. $HX, $HY, and $HZ for X, Y, and Z-axis homing. The full homing 
// cycle is still invoked by the $H command. This is disabled by default. It's here only to address
// users that need to switch between a two-axis and three-axis machine. This is actually very rare.
// If you have a two-axis machine, DON'T USE THIS. Instead, just alter the homing cycle for two-axes.
 #define HOMING_SINGLE_AXIS_COMMANDS // Default disabled. Uncomment to enable._**

wsjlmq, 2021-05-18:
Although the command **_$HX_** can be recognized, the **_X axis_** movement is completely wrong.(Others have the same problem)   I tried to modify the parameters, but I couldn't solve it.Now I'm trying this command **_$HY_**

wsjlmq, 2021-05-18:
As soon as it touches the Hard limit(Except for **_$H_**), it will stop. Is there any way to contact the Hard limit alarm?

langwadt, 2021-05-18:
no, line 75-76.  change them to: 

#define HOMING_CYCLE_0 (1<<X_AXIS)             
//#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.

wsjlmq, 2021-05-18:
> no, line 75-76. change them to:
> 
> #define HOMING_CYCLE_0 (1<<X_AXIS)
> //#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS)) // OPTIONAL: Then move X,Y at the same time.

[image]

wsjlmq, 2021-05-18:
> I think this is the place you said
> **_// Enables single axis homing commands. $HX, $HY, and $HZ for X, Y, and Z-axis homing. The full homing // cycle is still invoked by the $H command. This is [...]

By changing this place, the command _$HX_ can be identified.
However, the  distance to find the limit is not correct. It will only move the distance set by this command $27 （homing switch pull-off distance ）each time.The distance to find the limit should have been determined by $130.
[image]

langwadt, 2021-05-18:
pull-off is how far it moves away from the switch after it has found it, not how far to move to find it

wsjlmq, 2021-05-20:
> pull-off is how far it moves away from the switch after it has found it, not how far to move to find it

Thank you. I see what you mean.
But the problems I encountered before are exactly as described above. Fortunately, after I burned the firmware many times, it returned to normal.
But I still don't know what caused it