grbl/grbl #1724: Stepper not moving, using grbl en cnc shield

Jagvir0603, 2021-05-24:
I have been working on repairing an old cnc, have tried alot but nothing seems to help.
So today I setup an aduino, breadboard with drv8825 and a random [stepper](https://www.mitsumi.co.jp/latest/Catalog/compo/motor/m35sp11nk_e.html) I got of a friend.
With the following code I got the stepper moving see video.
https://user-images.githubusercontent.com/36311413/119392877-955c3680-bcd0-11eb-81d2-927aef973dc7.mp4
Code used: `/*Example sketch to control a stepper motor with A4988/DRV8825 stepper motor driver and Arduino without a library. More info: https://www.makerguides.com */

// Define stepper motor connections and steps per revolution:
#define dirPin 2
#define stepPin 3
#define stepsPerRevolution 200

void setup() {
  // Declare pins as output:
  pinMode(stepPin, OUTPUT);
  pinMode(dirPin, OUTPUT);
}

void loop() {
  // Set the spinning direction clockwise:
  digitalWrite(dirPin, HIGH);

  // Set the spinning direction clockwise:
  digitalWrite(dirPin, HIGH);

  // Spin the stepper motor 5 revolutions fast:
  for (int i = 0; i < 5 * stepsPerRevolution; i++) {
    // These four lines result in 1 step:
    digitalWrite(stepPin, HIGH);
    delayMicroseconds(500);
    digitalWrite(stepPin, LOW);
    delayMicroseconds(500);
  }

  delay(1000);

  // Set the spinning direction counterclockwise:
  digitalWrite(dirPin, LOW);

  //Spin the stepper motor 5 revolutions fast:
  for (int i = 0; i < 5 * stepsPerRevolution; i++) {
    // These four lines result in 1 step:
    digitalWrite(stepPin, HIGH);
    delayMicroseconds(500);
    digitalWrite(stepPin, LOW);
    delayMicroseconds(500);
  }

  delay(1000);
}`

On the breadboard I measured the [drv8825](https://lastminuteengineers.com/wp-content/uploads/arduino/DRV8825-Stepper-Motor-Driver-Pinout.png):
EN, RST, SLP 5v and VMOT 18v

So when I put the cnc shield v3 on top of the arduino and plugin the power supply and try moving the stepper with universal g-code sender nothing happens. When I check the EN, RST, SLP pins they all seem fine but the VMOT is about 0.2v.

Does anyone have an idea what I am doing wrong?

the555timer, 2021-09-15:
The direction and step pins could be changed up on the cnc shield

the555timer, 2021-09-15:
the driver has a different pinout then what the cnc shield is designed for

JagvirSB, 2021-09-15:
The issue has been resolved it turned out the fuse in the CNC shield was burned out. 

Everything worked just fine with a new CNC shield.

closed 2021-09-15 by JagvirSB