grbl/grbl #1728: missalignment problem

sbw12345, 2021-08-06:
I hope someone with knowlede of grbl and cnc machines can help me. I can't get my 3018 cnc router to cut shapes on the same origins. For example if I load a program to cut a 40mm circle, set origins and run the program then load  a new program to cut  50 mm circle on the same origins I end up with two circles the correct size but the distance between the circles is not even.  The difference between the smallest and larges gap can be about 1 mm.  The x/y values in the program would suggest that the two circles should be consentric but they are not.

Has anyone experience of an issue like this or have any suggestions on what I could try?

jahnj0584, 2021-08-06:
Your motors are losing steps/positions. Reduce acceleration, top speed,
cutting force or add lubrication (to the screws) until it no longer
happens.

sbw12345, 2021-08-06:
Thanks for taking the time to help.  If it was missing steps I would expect the circle to be misshapen and they are perfect circles and the exact size the program should cut. The only problem is they are not cut in the correct place.  Neither circle has the origins point at its centre.

Norwegian-Gadgetman, 2021-08-06:
Can you post some example code, and also a picture of the result where you also mark where [0,0] is, and the assumed center should be?

Does your machine have endstops?

Trygve

sbw12345, 2021-08-07:
Hi. I have tried to attach the gcode filesand an image of the results when they are run. Any ideas would be appreciated.

I am not sure what you mean by "endstops". If that is limit swithes then yes, if not could you please expalin further.
Thanks

[square motif 40mm.txt](https://github.com/grbl/grbl/files/6949562/square.motif.40mm.txt)
[cut out 80mm Dd.txt](https://github.com/grbl/grbl/files/6949552/cut.out.80mm.Dd.txt)
[decor ring 70mm D.txt](https://github.com/grbl/grbl/files/6949553/decor.ring.70mm.D.txt)
[image]

jahnj0584, 2021-08-07:
If your square measures square, then i bet the lost motion comes from the
rapids between shapes. Rerurn this test with 50% rapids (or equal to cut
rate) to eliminate this option!

sbw12345, 2021-08-07:
Yes the squares and circles are the correct size, just ot in the correct position. I am not sure what you mean by "rapids between shapes".  Each shape is run as a seperate program starting from the origins position. Are you suspecting the G0 command at the start of each program is where it is loosing count?

jahnj0584, 2021-08-07:
Yes that's it. Your machine is either binding or losing steps when it's
trying to reach or run its maximum speed(rapids!)

sbw12345, 2021-08-07:
OK thanks I wil try that and let you know what happens.

sbw12345, 2021-08-08:
Hi. I tried adjusting the max speed and acceleration, but it did not make any difference. I edited the programs to change the G0 command to a G1commands at F200. There was no change in the results. If there are missing steps at the begining of the program it looked like there was another cause.

I tried replacing the spinde motor with the original. When I bought the machine I purchased a faster motor as an optional extra. This solved the issue and the programs were all concentric. It looks like the faster motor was causing the probem.

Thanks for you help tracking down the cause.

closed 2021-08-09 by sbw12345