grbl/grbl #1729: Stoping job without losing coordinates

Discorez, 2021-08-08:
Hi!
Which command I can stop the job immediately and clear the buffer without losing the machine position and resetting the coordinates.
I tried first to send the command "!" to stop and hold the axes, and next the command 0x18, but an Alarm occurs and the coordinates are reset.
Perhaps there is some other way?

tmpvar, 2021-08-08:
I think you need to wait for the machine state to be `Idle` before performing the reset.

Discorez, 2021-08-08:
Thanks for youre reply, tmpvar.
I agree. But, after sending the command "!" the machine is in a "hold" state. 
How to set it to "idle" state?

tmpvar, 2021-08-08:
Oh, sorry I meant `Hold`!  Are you resetting immediately after the hold command or do you wait for the machine's status to change before resetting?