grbl/grbl #1734: End Stops connection

gioiellerianotini, 2021-09-03:
I can't connect the end stops on the CNC shield. I don't find a guide to be able to do it correctly. Could you help me please? Thank you

[image]

[image]

xsnoopy, 2021-09-15:
You should at least add some links to the hardware you asking about.

gioiellerianotini, 2021-09-15:
thanks, i built this CNC with dremel and i would like to add end stops

[image]

DanWarure, 2023-01-23:
Hello, Please I need your help with the CNC GRBL libraries and firmwares. I tried downloading from different sites but it is not working

gioiellerianotini, 2023-01-23:
your hardware?

DanWarure, 2023-01-23:
Arduino UNO
CNC shield
[image]
[image]
[image]
[image]