grbl/grbl #1738: UGS Error: Grbl has not finished booting

jasbneal, 2021-09-18:
Hi everyone 👋🏾

I'm very new to the Arduino, Grbl + [UGS](https://github.com/winder/Universal-G-Code-Sender) communities so I appreciate your patience in advance! 

I'm in the process of building a syringe pump to experiment with the electrospinning of nanofibers and using an Arduino Uno R3 board (NOT a Chinese clone, board has a ATMEGA328 chip and no CH340 chip) along with a [DRV8825 driver carrier](https://lastminuteengineers.com/drv8825-stepper-motor-driver-arduino-tutorial/), CNC shield v3 and NEMA 17 stepper motor. I did the current limit adjustment and attached the heat sink for the DRV8825 carrier + I left M0, M1 and M3 disconnected so that the driver operates in full-step mode so I hope I've avoided the common pitfalls.. 🤞🏾

I am able to successfully upload [Grbl v1.1g](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#quick-guide-to-setting-up-your-machine-for-the-first-time) to the Arduino Uno board however, when I connect via UGS and attempt to send any commands, I receive an error message saying "Grbl has not finished booting." I've triple checked the port and the baud is set to 115200. 

I'm even more confused because I was able to successfully submit the commands in UGS to test the motor rotation before uploading the code to the Arduino board (below). When the motor didn't rotate after the code was uploaded, I returned to UGS to test again and encountered this error. 

Help? 

[image]

````
/*Example` sketch to control a stepper motor with A4988/DRV8825 stepper motor driver and Arduino without a library. More info: https://www.makerguides.com */

// Define stepper motor connections and steps per revolution:
#define dirPin 2
#define stepPin 3
#define stepsPerRevolution 200

void setup() {
  // Declare pins as output:
  pinMode(stepPin, OUTPUT);
  pinMode(dirPin, OUTPUT);
}

void loop() {
  // Set the spinning direction clockwise:
  // HIGH = CW; LOW = CCW
  digitalWrite(dirPin, HIGH);

  // Spin the stepper motor 1 revolution slowly:
  for (int i = 0; i < stepsPerRevolution; i++) {
    // These four lines result in 1 step:
    digitalWrite(stepPin, HIGH);
    delayMicroseconds(2000);
    digitalWrite(stepPin, LOW);
    delayMicroseconds(2000);
  }

  delay(1000);

  // Set the spinning direction counterclockwise:
  digitalWrite(dirPin, LOW);

  // Spin the stepper motor 1 revolution quickly:
  for (int i = 0; i < stepsPerRevolution; i++) {
    // These four lines result in 1 step:
    digitalWrite(stepPin, HIGH);
    delayMicroseconds(1000);
    digitalWrite(stepPin, LOW);
    delayMicroseconds(1000);
  }

  delay(1000);

  // Set the spinning direction clockwise:
  digitalWrite(dirPin, HIGH);

  // Spin the stepper motor 5 revolutions fast:
  for (int i = 0; i < 5 * stepsPerRevolution; i++) {
    // These four lines result in 1 step:
    digitalWrite(stepPin, HIGH);
    delayMicroseconds(500);
    digitalWrite(stepPin, LOW);
    delayMicroseconds(500);
  }

  delay(1000);

  // Set the spinning direction counterclockwise:
  digitalWrite(dirPin, LOW);

  //Spin the stepper motor 5 revolutions fast:
  for (int i = 0; i < 5 * stepsPerRevolution; i++) {
    // These four lines result in 1 step:
    digitalWrite(stepPin, HIGH);
    delayMicroseconds(500);
    digitalWrite(stepPin, LOW);
    delayMicroseconds(500);
  }

  delay(1000);
}
````

AlexHolden, 2021-09-18:
If I understand correctly, you installed the grbl firmware on the Uno and it worked correctly. Then you replaced grbl with a simple Arduino stepper driver example sketch, and because the Uno was no longer running grbl, UGS couldn't talk to it any more.

jasbneal, 2021-09-18:
@AlexHolden You're right, I completely missed the obvious! 🤦🏾‍♀️ Also figured out that I needed to enable Pin 8 in order for the motor to work with the code so I added `#define enablePin 8`  and `pinMode(enablePin, OUTPUT);` `digitalWrite(enablePin, LOW);` to the `void setup()` function.

Thank you x100! 🙌🏾

closed 2021-09-18 by jasbneal