grbl/grbl #1741: Question on GRBL

Orion256, 2021-09-27:
Good day. I have one question about GRBL. Is it possible to add a position sensor to this firmware?

ImranKhan203, 2021-09-28:
Hello,

Please help me out.
**I am using GRBL 0.9 with the help of bCNC software and I want to attach Tower Lamp/Stack Light on our CNC machine.**
I would like to add 2 outputs on our machine to indicate if we’re running the program (green light) or if we’re in pause (called ‘hold’ in the program / red light). For that, I have to work on bCNC that is the software used in all our machines.

So, when the program is running, the green light has to be lighted, and if we’re in pause, the red light has to be lighted. The orange one will probably be used for something else in the future. 
To do that, I’ll have to detect the state of the program and send it to the lights (2 outputs).

**Please tell me in which file of bCNC and which type of program I have to use?**

ImranKhan203, 2021-09-28:
**Here is the file which I am using on Python 2.7.16 :**
[bCNC01.ZIP](https://github.com/grbl/grbl/files/7245626/bCNC01.ZIP)

**I want to attach this type of Tower Lamp/Stack Light :**
[image]

**Please suggest me in which file of bCNC and which type of program I have to use?**

terjeio, 2021-09-29:
@Orion256 There is not much room to add anything to this version of grbl. Is it a sensor for positional feedback?

@ImranKhan203 I guess you should have started a new issue instead og hijacking this... Anyway you may find [this project](https://github.com/5ocworkshop/grblhal-rgb-plugin/blob/main/rgb_plugin_readme.txt) of interest.

Orion256, 2021-09-29:
> @ Orion256 В эту версию grbl не так уж много места, чтобы что-то добавить. Это датчик позиционной обратной связи?
> 
> @ ImranKhan203 Думаю, вам следовало бы начать новый выпуск, а не угонять это ... В любом случае, [этот проект](https://github.com/5ocworkshop/grblhal-rgb-plugin/blob/main/rgb_plugin_readme.txt) может вас заинтересовать.

Yes. As a goal of achieving the writing of the diploma, the head instructed me. That I would describe the principle of modifying this living for adding a positional feedback sensor. So I wanted to first clarify with you Terjeio whether this is possible or not

terjeio, 2021-09-30:
For this version of grbl, IMO not possible as there is no room in the flash. The [Mega version](https://github.com/gnea/grbl-Mega/issues) has plenty so, again IMO, maybe - this since both versions are for a relatively slow 8-bit MCU pushed to its limits. You may have better luck with one of the 32-bit ports.

Orion256, 2021-09-30:
> For this version of grbl, IMO not possible as there is no room in the flash. The [Mega version](https://github.com/gnea/grbl-Mega/issues) has plenty so, again IMO, maybe - this since both versions [...]

Thanks for the answer. Alas, according to the assignment that they gave me for my diploma. I need to implement a control system for moving along three planes with a milling machine on Arduino Uno. P.S. Sorry its my bad english

closed 2021-10-01 by Orion256