grbl/grbl #1756: motor lockup when homing

jacobv95, 2021-12-03:
I have connected 2 end stops for each axis. Both the movements and the end stops works fine when jogging. But when i home ($H) all motors lock up and nothing happens - except for a high pitch noise from the motors.

All end stops are wired normally open

$0 = 3    (Step pulse time, microseconds)
$1 = 0    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 0    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.010    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 250.000    (Homing locate feed rate, mm/min)
$25 = 25.000    (Homing search seek rate, mm/min)
$26 = 5    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 200.000    (X-axis travel resolution, step/mm)
$101 = 200.000    (Y-axis travel resolution, step/mm)
$102 = 200.000    (Z-axis travel resolution, step/mm)
$110 = 250.000    (X-axis maximum rate, mm/min)
$111 = 250.000    (Y-axis maximum rate, mm/min)
$112 = 250.000    (Z-axis maximum rate, mm/min)
$120 = 25.000    (X-axis acceleration, mm/sec^2)
$121 = 25.000    (Y-axis acceleration, mm/sec^2)
$122 = 25.000    (Z-axis acceleration, mm/sec^2)
$130 = 0.000    (X-axis maximum travel, millimeters)
$131 = 0.000    (Y-axis maximum travel, millimeters)
$132 = 0.000    (Z-axis maximum travel, millimeters)

DM-Anthony, 2022-02-05:
$2 and $3 both cannot be 0

jacobv95, 2022-02-05:
$2 is the step port invert mask. It is used to invert the step pulses from normal-low to normal-high. My cnc works fine in all aspects but the hominig. Therefore, i dont think this is the problem.

$3 is the direction port invert mask. It simply inverts the positive direction of an axis. Again, my cnc works fine in all apsects but homing. So i dont think this is the problem either.

Why cant $2 and $3 both be equal to 0? This would be the case if all your motors works with a normal-low signal, and all axis have the correct positive direction. Or am i mistaken ?

jacobv95, 2022-02-06:
Regarding soft limits and maximum travel. If soft limits are enabled, then GRBL will check if any new command will exceed the maximum travel of the machine. I have disabled the soft limit, as I have hard limits installed.

jahnj0584, 2022-02-06:
Soft limits are to be used WITH homing switches. Stops you from running
into the opposisite ends.

jacobv95, 2022-02-06:
I will try to enable the soft limits. I will get back to you with the results

jacobv95, 2022-02-06:
I tried enabling the soft limits ($20) and setting the maximum travel to a large value (700x700x400). This did not fix the homing issue. 

But i was not able to run a G91 command with the soft limits on. I get the following error message: "Error while processing response <An error was detected while sending 'g91 y20': (error:9) G-code commands are locked out during alarm or jog state. Streaming has been pause". I even tried to reset zero before running the G91 command, in order to keep the machine within the maximum travel distance. This did not help.

Current machine settings:
$0 = 3    (Step pulse time, microseconds)
$1 = 0    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 0    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.010    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 250.000    (Homing locate feed rate, mm/min)
$25 = 25.000    (Homing search seek rate, mm/min)
$26 = 5    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 50.000    (X-axis travel resolution, step/mm)
$101 = 50.000    (Y-axis travel resolution, step/mm)
$102 = 200.000    (Z-axis travel resolution, step/mm)
$110 = 5000.000    (X-axis maximum rate, mm/min)
$111 = 5000.000    (Y-axis maximum rate, mm/min)
$112 = 1000.000    (Z-axis maximum rate, mm/min)
$120 = 500.000    (X-axis acceleration, mm/sec^2)
$121 = 500.000    (Y-axis acceleration, mm/sec^2)
$122 = 500.000    (Z-axis acceleration, mm/sec^2)
$130 = 700.000    (X-axis maximum travel, millimeters)
$131 = 700.000    (Y-axis maximum travel, millimeters)
$132 = 400.000    (Z-axis maximum travel, millimeters)

MohamedEyssa, 2022-04-21:
> I tried enabling the soft limits ($20) and setting the maximum travel to a large value (700x700x400). This did not fix the homing issue.
>
> But i was not able to run a G91 command with the soft limits [...]

> I have connected 2 end stops for each axis. Both the movements and the end stops works fine when jogging. But when i home ($H) all motors lock up and nothing happens - except for a high pitch noise from the motors.
> 
> All end stops are wired normally open
> 
> GRBL 1.1 $0 = 3 (Step pulse time, microseconds) $1 = 0 (Step idle delay, milliseconds) $2 = 0 (Step pulse invert, mask) $3 = 0 (Step direction invert, mask) $4 = 0 (Invert step enable pin, boolean) $5 = 0 (Invert limit pins, boolean) $6 = 0 (Invert probe pin, boolean) $10 = 0 (Status report options, mask) $11 = 0.010 (Junction deviation, millimeters) $12 = 0.010 (Arc tolerance, millimeters) $13 = 0 (Report in inches, boolean) $20 = 0 (Soft limits enable, boolean) $21 = 1 (Hard limits enable, boolean) $22 = 1 (Homing cycle enable, boolean) $23 = 0 (Homing direction invert, mask) $24 = 250.000 (Homing locate feed rate, mm/min) $25 = 25.000 (Homing search seek rate, mm/min) $26 = 5 (Homing switch debounce delay, milliseconds) $27 = 5.000 (Homing switch pull-off distance, millimeters) $30 = 1000 (Maximum spindle speed, RPM) $31 = 0 (Minimum spindle speed, RPM) $32 = 0 (Laser-mode enable, boolean) $100 = 200.000 (X-axis travel resolution, step/mm) $101 = 200.000 (Y-axis travel resolution, step/mm) $102 = 200.000 (Z-axis travel resolution, step/mm) $110 = 250.000 (X-axis maximum rate, mm/min) $111 = 250.000 (Y-axis maximum rate, mm/min) $112 = 250.000 (Z-axis maximum rate, mm/min) $120 = 25.000 (X-axis acceleration, mm/sec^2) $121 = 25.000 (Y-axis acceleration, mm/sec^2) $122 = 25.000 (Z-axis acceleration, mm/sec^2) $130 = 0.000 (X-axis maximum travel, millimeters) $131 = 0.000 (Y-axis maximum travel, millimeters) $132 = 0.000 (Z-axis maximum travel, millimeters)

Hello!
Did you find a Solution to this Homing Issue?
As I face it too now.

MohamedEyssa, 2022-04-21:
> I tried enabling the soft limits ($20) and setting the maximum travel to a large value (700x700x400). This did not fix the homing issue.
>
> But i was not able to run a G91 command with the soft limits [...]

> I have connected 2 end stops for each axis. Both the movements and the end stops works fine when jogging. But when i home ($H) all motors lock up and nothing happens - except for a high pitch noise [...]

Hello Jacob!
Did you find a Solution to this Homing Issue?
As I face it too now.

jacobv95, 2022-04-21:
No, i have not fixed this error yet