grbl/grbl #1763: Jerky Movement

jmb-555, 2021-12-25:
There have been recent problems with a GRBL machine. Multiple parts have been made with the machine with no problems. The configuration is correct. The problem is the machine is jerking/acting erratically. The steps/mm are correct and acceleration values are appropriate. Its moving very slowly and not being overloaded with too many gcode commands to the point that its overwhelmed. The strange thing is that when the relay that is plugged into the coolant pin is unplugged, the machine runs smooth as silk and the instant the coolant relay is plugged back in it becomes unusably jerky again. An arbitrary load was put on the coolant pin and the same jerky movement was observed but the jerks were no where near as bad as when the actual relay was plugged in. I am posting here because it seems to be isolated to the control board and I have not written this software and there is difficulty explaining how this could have happened. The relay that is being used is part of a module that has appropriate loop back protection when the relay fires off and even if it didnt it seems more likely to blow out the coolant pin and not wreck the entire robot like that but I'm not sure so I'm asking. Any help would be greatly appreciated and thanks again for posting up this code it is an impressive piece of software. 

Cheers and merry Christmas

jmb-555, 2021-12-25:
Video of problem shown here. As soon as the relay clicks off the machine runs smooth. Having trouble explaining this one...

https://user-images.githubusercontent.com/66437840/147389304-1474559b-4c5e-4b11-bae5-35d4511b1de1.mp4

langwadt, 2021-12-25:
the current to drive the relay is overloading the 5V supply?

jmb-555, 2021-12-26:
It seems to have been a bad/failing stepper motor driver. 

Switched the micro controller
No fix
Reflashed the firmware.
No fix
Checked the current on the stepper drivers
Were correct. No fix
Checked the configuration
Correct. No Fix
Switched around the stepper motors
The one that was jittery before was no longer jittery in a different position
Replaced the stepper motor driver on the jerky axis
Problem went away. 

Checked current draw of arduino. Arduino is taking in 130ma. GRBL shield is taking power from 12v supply. The 5v pin on the GRBL shield is probably from the Arduino regulator. From memory, the arduino is good for up to about 500ma. The relay module is consuming about 70-90ma when closed. It does not make sense that when switching on the relay the machine would go wonky because it is well under its 500ma maximum draw and the 12v GRBL supply is well over powered as well.

At this point, the motor driver replacement has fixed the problem but it is entirely a mystery how plugging in a relay would cause it to go jerky unless there is something im not seeing.

closed 2021-12-26 by jmb-555

langwadt, 2021-12-26:
The 5V is probably straight from the USB port so the current available depends, ans some USB cables have very thing wires. What drivers are you using? maybe, for some reason, the 5V end up being powered from 12V by one of the drivers internal regulator and the additional load puts it in thermal protection

dogserious, 2024-04-19:
You should try joining all the grounds together!