grbl/grbl #1772: Sometimes z axis won't going up

alfarosi, 2022-02-24:
My z axis sometimes won't going up. So if I'm doing 2d contour, z axis move from z0.5 to z5 and then to z-1. But sometimes it wont go to z5 dispite it was written z5 on the g code. And its destroying my works because now it has to run to 5.5mm depth instead of 0.5. any solution?

c3ntaur, 2022-02-28:
I had a similar problem, on a plasma cutting table though, where the z axis was just moving if I gave it a big distance to move. My problem has been that the spindle of the z movement was so hard to turn and had some "flex" (?) in it that it didn't stay in the position it should when moved small steps. Solution for me was always have current running through the stepper to provide a holding torque. Might have nothing to do with your setup. But maybe it's worth checking...

alfarosi, 2022-10-13:
For my problem, it turns out my stepper driver get overheated and now completely dead. Replacing new driver and adding little fan and the problem is gone now.

closed 2022-10-13 by alfarosi