grbl/grbl #1775: Dynamic power mode M4 issue

alexbn71, 2022-03-05:
I have a strange behaviour on both 1.1f and 1.1h version.  The following streams are generated by lightburn:

`Starting stream
G00 G17 G40 G21 G54
G90
M4
G0 X504Y269
Layer C00
G1 X503.203S1000F6000
G1 X561
G1 S0
M5
G90
G0 X0 Y0
M2
Stream completed in 0:02

Starting stream
G00 G17 G40 G21 G54
G90
M4
G0 X504Y269
Layer C00
G1 X503.203S1000F1200
G1 X561
G1 S0
M5
G90
G0 X0 Y0
M2
Stream completed in 0:03`

The problem is that the first stream with F6000 generate a PWM signal with a duty cycle lower than the second stream with F1200! 
As a consequence, the slower it goes, the more power is generated!

alexbn71, 2022-03-05:
Someone can point me in the code where the math for pwm is done based to the current feed rate?

alexbn71, 2022-03-06:
This issue has no sense! I’ve tried with a new MKS DLC 2.1 (a spare part I have) mounted on the bench only with a serial cable ad an oscilloscope hooked to D11/TTL port … uploaded the new grbl 1.1h… resetted all parameters to default with $RST=* then enabled only homing and laser mode ($22,$23). If I use the “fire” button in Lightburn… changing the power from 0 to 20% the PWM signal is correct!.. the duty cycle is coherent with the power value.
If I try to send an M4 follewed by G1 X503.203S1000F6000 or G1 X503.203S1000F1200 the command with the lowest feed rate (F1200) value produce a greater duty cycle on the PWM :frowning:

Going from a duty cycle of about 5% with F6000 to almost 50% with F1200

ggallant571, 2022-03-24:
It could be related to acceleration. Do you get expected PWM with M3 instead of M4? What is yours max PWM value? What is "Layer C00"?