grbl/grbl #1776: After succesfull home the machine first move is to Z=0, this triggers alarm

i-g-g-y, 2022-03-06:
Hello.

I work with F360 and very first move in code is:

`G90 G94
G17
G21
(When using Fusion 360 for Personal Use, the feedrate of)
(rapid moves is reduced to match the feedrate of cutting)
(moves, which can increase machining time. Unrestricted rapid)
(moves are available with a Fusion 360 Subscription.)
G28 G91 Z0
G90

(Face12)
T1
S1000 M3
`

This of course triggers my Z axis endpoint switch and raises an alarm.

How can I put my Z switch at Z=1mm, so a move to Z=0 does not trigger the switch?

langwadt, 2022-03-06:
define HOMING_FORCE_SET_ORIGIN (line 87 in config.h) it'll set the machine position to 0,0,0 after the switch pull off

alternatively just remove the lines with G28 (you can also turn them off in the post processor)

i-g-g-y, 2022-03-06:
#define HOMING_FORCE_SET_ORIGIN

did not work.

langwadt, 2022-03-06:
are you sure you changed, build, and flashed the right code?

what is the machine position after homing?

i-g-g-y, 2022-03-06:
Yes, I am sure, as I used the same process to disable the spindle control, so I can have all 3 switches.

langwadt, 2022-03-06:
what is the machine position after homing?

i-g-g-y, 2022-03-06:
Man I just damaged my machine. I have to change one of my switches. I have to disassemble the complete machine.

i-g-g-y, 2022-03-06:
I fixed my machine.

Here is the screen after homing:
[image]

And the soft limits:
[image]

Config for homing:
[image]

i-g-g-y, 2022-03-06:
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 4    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 16    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 100.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 200    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 400.000    (Y-axis travel resolution, step/mm)
$102 = 161.000    (Z-axis travel resolution, step/mm)
$110 = 1000.000    (X-axis maximum rate, mm/min)
$111 = 1000.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 277.000    (X-axis maximum travel, millimeters)
$131 = 190.000    (Y-axis maximum travel, millimeters)
$132 = 160.000    (Z-axis maximum travel, millimeters)
ok

langwadt, 2022-03-06:
ah you are homing in the reverse direction, then I don't know if HOMING_FORCE_SET_ORIGIN will work then, never tried. 

you could change, line 302-304 in limits.c, removing the pull off

i-g-g-y, 2022-03-06:
Don't I need this pull-off?

langwadt, 2022-03-06:
the pull off is part of the homing cycle, but those lines set the position after the homing is all done. i.e. with you 5mm pulloff on Z it sets the Z position to -5mm after homing is done, what you want and imo what makes most sense it to set Z = 0mm after homing is done

i-g-g-y, 2022-03-06:
OK, can you give me the exact syntax to change? I have GRBL 1.1h and my line numbers do not correspond to the ones you are giving me.

i-g-g-y, 2022-03-06:
[image]

langwadt, 2022-03-06:
ok, then you should really be in https://github.com/gnea/grbl/ not this one

anyway:    remove the HOMING_FORCE_SET_ORIGIN define, that won't work with reverse homing 
change line 382 in limits.c from:
`set_axis_position = lround(-settings.homing_pulloff*settings.steps_per_mm[idx]);`
to:
`set_axis_position = 0`

then you Z position will be 0mm after moving away from the switch, instead of -5mm

i-g-g-y, 2022-03-06:
Still the same??

[image]

[image]

langwadt, 2022-03-06:
is it Z or X/Y that gets triggered?

what G28 Z0 does is move to Z=0 and the to the position stored for G28 if you haven't stored a position that is 0,0,0
 
move the spindle to a convenient safe place and store that position with G28.1

i-g-g-y, 2022-03-06:
It is Z axis.

langwadt, 2022-03-06:
what is the Z position after homing is done? it should 0 and not touching the switch

i-g-g-y, 2022-03-06:
Nope. How do I figure out where is my code coming from? Looks like I am modifying code that is not used? I am completely confused now.

[image]

i-g-g-y, 2022-03-06:
> ok, then you should really be in https://github.com/gnea/grbl/ not this one

Sorry, what?

i-g-g-y, 2022-03-06:
> ok, then you should really be in https://github.com/gnea/grbl/ not this one

I really do not understand what you ment?

langwadt, 2022-03-06:
https://github.com/grbl/grbl/ is the repo for older versions of grbl, newer version like the one you use is on https://github.com/gnea/grbl/  https://github.com/grbl/grbl/wiki#grbl-v11-has-been-released-at-our-new-site-the-old-site-will-eventually-be-phased-out

I haven't build it in arduino, but I've heard of people editing the wrong files

https://github.com/grbl/grbl/wiki/Compiling-Grbl

under the heading "Advanced Users: " might help
 
when not sure if you are editing the right file,  add some random characters and verify that you get a compile error

i-g-g-y, 2022-03-06:
No dice. as soon I tell the machine to go to work zero it throws an error: 

[image]

This is how it looks like after homing:

[image]

langwadt, 2022-03-06:
work position can be anywhere you set it to, homing set machine coordinates, work coordinates are set realtive to that when you zeros at some postition where you want work zero 

the error is not that you hit the switch(hard limit), it is that you moved outside the machines defined limits (soft limit)

your $132 = 160.000 means you Z axis can be in the range -160mm to 0mm, you are at 0 in machines coordinates and 2.7mm work coordinates , G0Z5 tells it to move to 5mm in work coordinates, which would be 2.3mm in machine coordinates, that is clearly outside the 0 to -160mm range

you should move the spindle to where your work zero should be and zero all axis first

i-g-g-y, 2022-03-06:
That is the thing. If I modify the code, it wants to unscrew itself and have to do an emergency stop. I do set work coordinates. I will show tomorrow.

i-g-g-y, 2022-03-07:
Hello.

With this code modification it works, I do not know what I was doing wrong. As a feature, I would request the limit switches be put out of bound of the moving area of the machine.

[image]

i-g-g-y, 2022-03-07:
Here a picture, so you can see where does the progress go. I have to modify the y bracket, but it does work.

[image]