grbl/grbl #1777: Soft Limits false trigger

kthornbloom, 2022-03-07:
## If an operation works perfectly with soft limits off, why should I get an error when I turn the soft limits on?

----

First of all, thank you for this free, open source project! 👏 I have a machine set up as follows:

[image]

When I enable soft limits, they trigger immediately when running any job, **even though the job works fine with limits off**.  I suspect it's because my machine operates into the "positive" X + Y planes, but the gcode from Fusion has negative numbers. 

```
G90 G94
G17
G21
G90

(2D Contour1)
G54
G0 Z5 
X-125.776 Y168.349
```
☝️ it dies here: `[Error] An error was detected while sending 'X-125.776Y168.349': (ALARM:2) Soft limit alarm. G-code motion target exceeds machine travel`

[image]

Settings:
[image]

So again, why are the soft limits not receiving the absolute positioning or work coords that the milling operation gets?

jahnj0584, 2022-03-07:
There is no x-125 move displayed in your graphics?

kthornbloom, 2022-03-07:
Hi @jahnj0584 - I pasted the gcode in my post, up until the line that triggers the warnings: `X-125.776 Y168.349`