grbl/grbl #1786: Starting spindle with M3 doesn't work properly

FireWingHu, 2022-03-29:
Hey Guys,
could you please help understanding or solving this issue?

I use this config:
Arduino Uno with Grbl 1.1 firmware
Candle 1.1.17 running on win10 (I've also tried with 1.2.15b)

If I click on the start motor button at the spindle section in Candle, it doesn't launch the motor (and doesn't produce any PWM sign) until I click on one of the direction buttons for the stepper motors (regardless that is X,Y, or Z axis). 

If I send manually the M3 S1000 command, it answers an OK, but the motor doesn't react, but if I send a G1 X10 for example after that, it turns on the spindle immediatelly, so it behaves in the same way as in the previous example.

If I run a gcode file, it's the same.

Is it normal? I don't think so. :) Could you pls explain this? Thanks in advance,
P

langwadt, 2022-03-29:
for version 1.1 you should be asking here: https://github.com/gnea/grbl/

anyway, did you accidentally enable laser mode?

FireWingHu, 2022-03-30:
Sorry for misaddressing, and thanks for your advice, $32=0 (laser mode off) has done the thing! :)

closed 2022-03-30 by FireWingHu