grbl/grbl #1791: Suggestion for a new $ command

AntonioFromBrazil, 2022-04-24:
What about a new $33 command to say VARIABLE_SPINDLE = yes/no ? This would avoid the necessity to recompile & reload the Grbl firmware every time laser to spindle or vice-versa was changed...

langwadt, 2022-04-24:
why do you need to change it between spindle and laser?

AntonioFromBrazil, 2022-04-24:
My CNC accept both laser and spindle. For laser I use the power control through D11 (-Z). For spindle I use only on/off through spindle enable pin D12. So every time when I change tool from laser to spindle or spindle to laser I should change the variable_spindle definition in config.h and recompile/reload the grbl...

langwadt, 2022-04-25:
look at the code in spindle_control.c try USE_SPINDLE_DIR_AS_ENABLE_PIN

and unless you are using a very old grbl you should be at https://github.com/gnea/grbl/ not here

AntonioFromBrazil, 2022-04-25:
Great. It works but there is only one small problem. The D13 is put in active state for few seconds when the Arduino is turned on or reseted. But I think I can find a solution for this small inconvenience. Thanks for help.