grbl/grbl #1798: Stepperonline DM860T stepper drivers hissing but moving after half a second

bbuster2, 2022-06-17:
I'm sorry for my bad title but I couldn't think of a better one.

I am using StepperOnline DM860T drivers connected to a GRBL controller.
Previously I used cheap TB6600 drivers which worked instantly, but after upgrading I am having problems manually moving the motors, and by manually I mean with the controller but not in a program.

When I press the button to move the X-axis for example, the motors hiss for half a second and than move (at least if I move them enough mm), if I move them for example 10mm they only hiss without moving. When I move them for 100mm they his for half a second and then move for a remaining 70mm for example.

I just figured out that if I set the $1 command to 255 (Sets the time delay in milliseconds that GRBL will power the stepper motors after a motion command is complete. A setting of 255 tells the motors to stay powered on to hold position.) the motors work perfectly fine and respond immediately. The only problem is that the controller always sees that the motors are enabled and I cannot do anything because I need to wait till the motors are finished.

Also the motors are enabled all the time with this setting which is probably not a good thing.
The StepperOnline DM860T have a few settings (see attachement).

The only problem is I cannot find a GRBL setting to make the motors enable more quickly or have a delay after the enabling before trying to move.

How can I solve this?

PS: The controller I am using is a Makerbase MKS DLC32
[image]

jahnj0584, 2022-06-17:
Just set $1 to 255. Everyone did. Motors are meant to stay idle and hold a
detent position.

bbuster2, 2022-06-17:
I can’t because all the touchscreen actions say:

“warning:

please wait machine stop”

Also i feel it still misses steps or parts of the movements. When i set the z home and then move the z axis 100mm down and press the home button again. It goes up but more than it went down. So i think when going down it couldn’t move the 100mm but only 90 dor example