grbl/grbl #1800: error 9

RAMickelsen, 2022-06-29:
I am a newbie trying to get my brand new 3018 Prover to work. I am running Meshcam and Candle. I have candle set up properly with the comport selected and the status reading "idle". I have a gcode created in meshcam loaded. I have the XY and Z axis zeroed. But when I click "send", the spindle fires up for about one second and then stops. Candle is green lighted reading "run", but nothing happens after that. I have tried multiple times. The console says, reset to continue. When I do this and press unlock, the console gives me an error 9. (s6500 < error:9)   Please help.

Here are my settings:

S6500 < ok
$$ < $0=10
$1=25
$2=0
$3=2
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=1
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=10000
$31=0
$32=0
$100=761.905
$101=761.905
$102=800.000
$110=2000.000
$111=2000.000
$112=600.000
$120=10.000
$121=10.000
$122=10.000
$130=260.000
$131=158.000
$132=34.000
ok

RAMickelsen, 2022-06-29:
G21 < ok
M3 S6500 < ok
G0X0.00Y0.00Z2.54 < ok
G0X66.01Y29.41 < ok
G1Z-0.29F720.3 < ok
G1F1440.6 < ok
X66.34Y29.39 < ok
X88.50 < ok
X88.52Y29.41 < ok
Y50.40 < ok
X88.57Y50.79 < ok
X88.67Y51.18 < ok
X88.86Y51.57 < ok
X89.13Y51.96 < ok
X89.56Y52.35 < ok
X90.05Y52.62 < ok
X90.44Y52.76 < ok
X90.83Y52.81 < ok
X99.38 < ok
X99.60Y53.13
Y100.16
X99.38Y100.33
X91.22
X90.83Y100.34
X90.44Y100.40
X90.05Y100.54
X89.67Y100.73
X89.40Y100.94
[MSG:Reset to continue]

RAMickelsen, 2022-06-29:
I can see that I am not getting < ok for the last nine lines of the gcode. Could this be the problem? If so, how do I fix this?

jahnj0584, 2022-06-29:
My guess is 6500 is either above or below your setting for max or min
spindle speeds. #30 maybe?

RAMickelsen, 2022-06-29:
$30 = max spindle speed. Mine is set to 10000. 6500 is well below that. 
$31 = min spindle speed. Mine is set to 0. 
So my min-max spindle speed is 0 - 10000. A setting of 6500 in the gcode should be no problem, right?

RAMickelsen, 2022-06-29:
$30 = max spindle speed. Set at 10000
$31 is min spindle speed. Set at 0. 

So, 0 - 10000. 6500 is right in the middle of that range. Any other ideas?