grbl/grbl #1811: Endstops on rotary axis'

MechApe, 2022-11-06:
Hi, many apologies if there is an answer to this somewhere - I couldn't find it!

I have a custom 3 axis machine with 2 rotary axis' and 1 linear axis. My 2 rotary axis' can rotate infinitely (+ and -) . However, it would be really useful to have an end stop to help with repeatable datuming.

If I were to wire 2 inductive sensors into the endstops on my controller, would this enable grbl to home the axis' whilst ignoring the endstops when operating?

For example, I home the "X Axis", then move to +370degrees (passing the end stop). Would grbl get upset at an end stop error when crossing the 0 (360degree) mark?

Thanks,
Wayne

terjeio, 2022-11-06:
You want it like [in this build](https://github.com/grblHAL/core/blob/master/changelog.md#20221018) of grblHAL - it might be possible to backport the changes but taht would require a Mega due to memory constraints?

>  Would grbl get upset at an end stop error when crossing the 0 (360degree) mark?

Only if hard and possibly soft limits are enabled.

MechApe, 2022-11-07:
Awesome, thanks for the point in the right direction. I'll have a play soon and report back!