grbl/grbl #1823: Laser cutter missing G02 circles.  Repeatable.

bz01t4x, 2023-02-16:
Hello, people -

I have what I think is a good puzzle.

GRBL 1.1h, 'real' Arduino Uno R3, ZenToolworks 12x12 CNC, Nema 23 steppers, TB6560 driver board.  24V 5A supply.  Gantry has a LaserTree 10W laser.  Using Win10 with UGS v2.0.13.

I normally use LinuxCNC with above and it always works.  It's just become a pain to obtain/maintain old parallel port computers.  So am trying GRBL.  Have done multiple stipple cuttings (between 1 and 2 K stipples or circles per job).  Both using LinuxCNC and GRBL to compare performance/quality.  With identical gcode.  I noticed the GRBL run skipping stipple (circle) cuts.  It pauses for a very short time, maybe just get a laser blip, and just continue on it's path to the next.  Maybe five or ten percent behave like this.  At first I thought it might have been noise - I used different USB cables, different flavors of Uno, behavior did not change.  Then after running the same job so often noticed the missing stipples persisted across different runs of the same job.   I used LinuxCNC to identify the missing stipples (it highlights in Gcode when selecting a stipple in Axis) while studying the 'diff' between LinuxCNC and GRBL runs on the actual card stock.  Attached is a photo of the diff.  Left is LinuxCNC, right is GRBL.

[image]

After identifying a few of the missing stipples, I decided to extract a section of gcode to a more manageable number of stipples - eight or nine.  There were two missing stipples within bounded correct stipples.  When just running the 8 stipple gcode job, both missing ones were created correctly.  But the LAST of the eight never did.  For fun, I added a ninth.  One through 7 correctly cut, the 8th and 9th did not, although the X, Y movements (X,Y paths) were correctly run for all stipples.

I've tried changing every GRBL config to debug and I'm out of ideas.  Below is the 9 stipple gcode.  It's only 40 lines.

/**********************************************************
G00 Z -76.0000
F250

G00 X 90.2618
G00 Y 6.3638
M3 S100
G02 X 90.2618 Y 6.3638 I 0.7774 J 0.7774

G00 X 89.4868
G00 Y 3.0351
M3 S100
G02 X 89.4868 Y 3.0351 I 0.7574 J 0.7574

G00 X 86.1371
G00 Y 2.1170
M3 S100
G02 X 86.1371 Y 2.1170 I 0.7774 J 0.7774

G00 X 87.0492
G00 Y 5.3627
M3 S100
G02 X 87.0492 Y 5.3627 I 0.7998 J 0.7998

G00 X 83.8751
G00 Y 5.1315
M3 S100
G02 X 83.8751 Y 5.1315 I 0.7774 J 0.7774

G00 X 82.7701
G00 Y 2.1492
M3 S100
G02 X 82.7701 Y 2.1492 I 0.7724 J 0.7724

G00 X 79.4909
G00 Y 1.9836
M3 S100
G02 X 79.4909 Y 1.9836 I 0.7350 J 0.7350

G00 X 76.1432
G00 Y 1.9572
M3 S100
G02 X 76.1432 Y 1.9572 I 0.7226 J 0.7226

G00 X 73.0000
G00 Y 2.0000
M3 S100
G02 X 73.0000 Y 2.0000 I 0.7226 J 0.7226

M5
M2
/**********************************************************

Below are my config settings:

/**********************************************************
**** Connected to COM5 @ 115200 baud ****
Grbl 1.1h ['$' for help]
>>> $$
$0 = 30    (Step pulse time, microseconds)
$1 = 1    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 2    (Step direction invert, mask)
$4 = 1    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 1    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 400.000    (Y-axis travel resolution, step/mm)
$102 = 400.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 30.000    (X-axis acceleration, mm/sec^2)
$121 = 30.000    (Y-axis acceleration, mm/sec^2)
$122 = 30.000    (Z-axis acceleration, mm/sec^2)
$130 = 325.000    (X-axis maximum travel, millimeters)
$131 = 325.000    (Y-axis maximum travel, millimeters)
$132 = 84.000    (Z-axis maximum travel, millimeters)
ok
/**********************************************************

Here is the UGS visual of the job.  Missing stipples circled in blue:

[image]

As I've said, I believe I've gone through a ridiculous number of config permutations trying to identify what could possibly be wrong.  Messed with $11 up and down, $12 up and down, $1, all the speeds/accelerations.  Laser mode on/off.  Everything.  I've also messed with the gcode over and over - using M4 instead of M3, slowing feeds down, putting explicit M5s in every block, putting explicit feeds in every block.  This just seems so simple.  As I've mentioned, LinuxCNC creates all these jobs without issue on same hardware.  Would LOVE to be able to transition to GRBL, at least for stippling.

Could someone please help?  Maybe run the above code and verify the last two circles don't turn (even if you don't have a laser).  I desperately need to know I'm not losing my mind.

Thanks

romanetz, 2023-02-17:
> image
> GRBL_UGS_FAIL

Hi, regarding LinuxCNC, we've developed a very cheap alternative to 
parallel port. Its performance is rather like to popular Mesa cards. 
More is here: 
https://forum.linuxcnc.org/27-driver-boards/44422-colorcnc-colorlight-5a-75e-5a-75b-as-fpga-controller-board
you can build it yourself from off-the-shelf LED control card. The 
process is rather trivial. Read this forum from 32-nd page.
Moreover, the Mesa firmware was ported to such card, the software and 
instructions are here: https://github.com/sensille/litehm2

17.02.2023 02:02, bz01t4x пишет:

terjeio, 2023-02-17:
This is an odd one, it even happens in my 32-bit port.
If I change [this code](https://github.com/grbl/grbl/blob/9180094b72821ce68e33fdd779278a533a7bf40c/grbl/motion_control.c#L103) to
`    if (-angular_travel >= ARC_ANGULAR_TRAVEL_EPSILON) { angular_travel -= 2*M_PI; }`
in my port the issue goes away for some strange reason. Too late at night now for me to dig into this now, but would be interesting to know if the change works for 8-bit MCUs as well.

[Here is the reason](https://github.com/grbl/grbl/blob/9180094b72821ce68e33fdd779278a533a7bf40c/grbl/config.h#L302-L310) for the code.

terjeio, 2023-02-18:
My comment above is bs, sorry for that. I changed the G54 offset to X-70 and Y-18 to run the test above faster, and the erratic behaviour moved to different circles. This kind of proves that lack of float precision is the culprit? How/where to fix this? Anyway - bedtime is long overdue so good night...

terjeio, 2023-02-18:
I believe I have solved the issue - for MCUs that has double support, AFAIK 8-bit Arduinos has not, they use float precison for double:

I changed [this code](https://github.com/grbl/grbl/blob/9180094b72821ce68e33fdd779278a533a7bf40c/grbl/motion_control.c#L93-L101) to

```
    double center_axis0 = (double)position[plane.axis_0] + (double)offset[plane.axis_0];
    double center_axis1 = (double)position[plane.axis_1] + (double)offset[plane.axis_1];
    double r_axis0 = -offset[plane.axis_0];  // Radius vector from center to current location
    double r_axis1 = -offset[plane.axis_1];
    double rt_axis0 = (double)target[plane.axis_0] - center_axis0;
    double rt_axis1 = (double)target[plane.axis_1] - center_axis1;
    // CCW angle between position and target from circle center. Only one atan2() trig computation required.
    float angular_travel = atan2(r_axis0 * rt_axis1 - r_axis1 * rt_axis0, r_axis0 * rt_axis0 + r_axis1 * rt_axis1);
```

This reduces the `angular_travel` error to 0 for the above program at the cost of slightly longer execution time.
I do not know how to solve this for MCUs without double support - perhaps someone with better float math knowledge than me can do it? [This paper](https://docs.oracle.com/cd/E19957-01/806-3568/ncg_goldberg.html) may help?

bz01t4x, 2023-02-18:
@terjeio, thanks for looking onto this.  I really appreciate the time you took.  But, I’m not quite sure I understand the nature of your problem/analysis solution.  If it was purely a float/double typing or arc resolution issue I would expect the failed arcs to fail regardless of where they exist in the G-code file.  One of the things I did to try to isolate the problem was take 2 circles failing consistently in one file and extract a block of G-code with them included located in the first eight blocks of another file.  The 2 circles previously that didn’t cut in the first file DID in the second.  But, the last block that DID print in the first file, didn’t in the second.  This tells me G-code block order is part of the problem.  If I didn’t know any better I would blame some kind of throughput issue and/or buffer overflow as the culprit.  I could imagine as circles are consecutively cut, excess time for each adding up and eventually overrunning the task, resulting in a loss of G02 command.  Hence the occasional laser flash I see after locating in X, Y but witnessing no G02 command executing.  If there were a way of pausing the stream from UGS every few circle blocks this could be tested, allowing all the blocks to fully execute before continuing I’d do it.  But I haven’t looked at the source and don’t have any context for how the code works yet.  I wasn’t expecting to have to go this deep.  Was hoping one of the OG developers could comment on a root cause before a fix is suggested.  Also, as mentioned previously, I did monkey around with the arc tolerance config.  My interpretation of what $12 does is trade processing time for arc accuracy.  Would have expected this to have had some effect if it really was throughput that was the issue.  But it didn’t - this is part of my utter bewilderment.  I guess the next step is try to figure out how the GRBL code is structured while waiting for folks to weigh in.  I’m guessing since GRBL is not a true Arduino sketch no debugging is possible?  No stepping through the GRBL code and inspecting registers, etc?

terjeio, 2023-02-18:
It has to do with float precision...
Here are logs from my initial tests before changing the code.

First run, no G54 offset:

```
[MSG:r: 1.099410e+00 dr: 0.000000e+00]
[MSG:t0: 2.026180e+01 t1: -1.163620e+01 c0: 2.103919e+01 c1: -1.085880e+01]
[MSG:at: 6.410689e-07 r0: -7.774000e-01 r1: -7.774000e-01 rt0: -7.773991e-01 rt1: -7.774000e-01]
[MSG:r: 1.071125e+00 dr: 0.000000e+00]
[MSG:t0: 1.948679e+01 t1: -1.496490e+01 c0: 2.024419e+01 c1: -1.420750e+01]
[MSG:at: -6.234200e-07 r0: -7.574000e-01 r1: -7.574000e-01 rt0: -7.574005e-01 rt1: -7.573996e-01]
[MSG:r: 1.099410e+00 dr: 0.000000e+00]
[MSG:t0: 1.613710e+01 t1: -1.588300e+01 c0: 1.691450e+01 c1: -1.510560e+01]
[MSG:at: 6.410689e-07 r0: -7.774000e-01 r1: -7.774000e-01 rt0: -7.773991e-01 rt1: -7.774000e-01]
[MSG:r: 1.131088e+00 dr: 0.000000e+00]
[MSG:t0: 1.704919e+01 t1: -1.263730e+01 c0: 1.784900e+01 c1: -1.183750e+01]
[MSG:at: -6.056623e-07 r0: -7.997999e-01 r1: -7.997999e-01 rt0: -7.998009e-01 rt1: -7.997999e-01]
[MSG:r: 1.099410e+00 dr: 0.000000e+00]
[MSG:t0: 1.387510e+01 t1: -1.286850e+01 c0: 1.465250e+01 c1: -1.209110e+01]
[MSG:at: 0.000000e+00 r0: -7.774000e-01 r1: -7.774000e-01 rt0: -7.774000e-01 rt1: -7.774000e-01]
[MSG:r: 1.092339e+00 dr: 0.000000e+00]
[MSG:t0: 1.277010e+01 t1: -1.585080e+01 c0: 1.354250e+01 c1: -1.507840e+01]
[MSG:at: 0.000000e+00 r0: -7.724000e-01 r1: -7.724000e-01 rt0: -7.723999e-01 rt1: -7.723999e-01]
[MSG:r: 1.039447e+00 dr: 0.000000e+00]
[MSG:t0: 9.490898e+00 t1: -1.601640e+01 c0: 1.022590e+01 c1: -1.528140e+01]
[MSG:at: 0.000000e+00 r0: -7.350000e-01 r1: -7.350000e-01 rt0: -7.349997e-01 rt1: -7.349997e-01]
[MSG:r: 1.021911e+00 dr: 0.000000e+00]
[MSG:t0: 6.143196e+00 t1: -1.604280e+01 c0: 6.865796e+00 c1: -1.532020e+01]
[MSG:at: 0.000000e+00 r0: -7.226000e-01 r1: -7.226000e-01 rt0: -7.226000e-01 rt1: -7.226000e-01]
[MSG:r: 1.021911e+00 dr: 0.000000e+00]
[MSG:t0: 3.000000e+00 t1: -1.600000e+01 c0: 3.722600e+00 c1: -1.527740e+01]
[MSG:at: 0.000000e+00 r0: -7.226000e-01 r1: -7.226000e-01 rt0: -7.226000e-01 rt1: -7.226000e-01]
[MSG:Pgm End]
```
Second run, with G54 offset:

```
[MSG:r: 1.099410e+00 dr: 0.000000e+00]
[MSG:t0: 9.026180e+01 t1: 6.363800e+00 c0: 9.103919e+01 c1: 7.141200e+00]
[MSG:at: 1.824583e-06 r0: -7.774000e-01 r1: -7.774000e-01 rt0: -7.773972e-01 rt1: -7.774000e-01]
[MSG:r: 1.071125e+00 dr: 0.000000e+00]
[MSG:t0: 8.948679e+01 t1: 3.035100e+00 c0: 9.024419e+01 c1: 3.792500e+00]
[MSG:at: -3.117099e-07 r0: -7.574000e-01 r1: -7.574000e-01 rt0: -7.574005e-01 rt1: -7.574000e-01]
[MSG:r: 1.099410e+00 dr: 0.000000e+00]
[MSG:t0: 8.613710e+01 t1: 2.117000e+00 c0: 8.691450e+01 c1: 2.894400e+00]
[MSG:at: 1.824583e-06 r0: -7.774000e-01 r1: -7.774000e-01 rt0: -7.773972e-01 rt1: -7.774000e-01]
[MSG:r: 1.131088e+00 dr: 0.000000e+00]
[MSG:t0: 8.704919e+01 t1: 5.362700e+00 c0: 8.784899e+01 c1: 6.162499e+00]
[MSG:at: 1.770401e-06 r0: -7.997999e-01 r1: -7.997999e-01 rt0: -7.997971e-01 rt1: -7.997999e-01]
[MSG:r: 1.099410e+00 dr: 0.000000e+00]
[MSG:t0: 8.387510e+01 t1: 5.131499e+00 c0: 8.465250e+01 c1: 5.908899e+00]
[MSG:at: 1.824583e-06 r0: -7.774000e-01 r1: -7.774000e-01 rt0: -7.773972e-01 rt1: -7.774000e-01]
[MSG:r: 1.092339e+00 dr: 0.000000e+00]
[MSG:t0: 8.277010e+01 t1: 2.149200e+00 c0: 8.354250e+01 c1: 2.921600e+00]
[MSG:at: 0.000000e+00 r0: -7.724000e-01 r1: -7.724000e-01 rt0: -7.723999e-01 rt1: -7.723999e-01]
[MSG:r: 1.039447e+00 dr: 0.000000e+00]
[MSG:t0: 7.949090e+01 t1: 1.983600e+00 c0: 8.022590e+01 c1: 2.718600e+00]
[MSG:at: -3.861654e-07 r0: -7.350000e-01 r1: -7.350000e-01 rt0: -7.350006e-01 rt1: -7.350000e-01]
[MSG:r: 1.021911e+00 dr: 0.000000e+00]
[MSG:t0: 7.614320e+01 t1: 1.957200e+00 c0: 7.686580e+01 c1: 2.679800e+00]
[MSG:at: -1.883507e-06 r0: -7.226000e-01 r1: -7.226000e-01 rt0: -7.226028e-01 rt1: -7.226001e-01]
[MSG:r: 1.021911e+00 dr: 0.000000e+00]
[MSG:t0: 7.300000e+01 t1: 2.000000e+00 c0: 7.372260e+01 c1: 2.722600e+00]
[MSG:at: -1.940583e-06 r0: -7.226000e-01 r1: -7.226000e-01 rt0: -7.226028e-01 rt1: -7.226000e-01]
[MSG:Pgm End]
```

`r` and `dr` values are for [radius and radius delta](https://github.com/grbl/grbl/blob/9180094b72821ce68e33fdd779278a533a7bf40c/grbl/gcode.c#L801-L804) for each of the circles. These are ok as the delta is 0 for all.

`t0`, `t1` are target [x,y in motion_control.c](https://github.com/grbl/grbl/blob/9180094b72821ce68e33fdd779278a533a7bf40c/grbl/motion_control.c#L86).

`c0`, `c1` are the [center_axis variables](https://github.com/grbl/grbl/blob/9180094b72821ce68e33fdd779278a533a7bf40c/grbl/motion_control.c#L93-L94).

`at` is the [angular travel variable](https://github.com/grbl/grbl/blob/9180094b72821ce68e33fdd779278a533a7bf40c/grbl/motion_control.c#L101). When this value goes negative the circle is ignored.

`r0`, `r1` and rt0, rt1 are the [radius vectors](https://github.com/grbl/grbl/blob/9180094b72821ce68e33fdd779278a533a7bf40c/grbl/motion_control.c#L95-L98).

Let us look at the last circle in the second log as it is ignored:

[MSG:t0:  __7.300000e+01__ t1: 2.000000e+00 c0: __7.372260e+01__ c1: 2.722600e+00]
[MSG:at: -1.940583e-06 r0: -7.226000e-01 r1: -7.226000e-01 rt0:  __-7.226028e-01__ rt1: -7.226000e-01]

This calulation is not precise:
rt0 = t0 - c0 -> 7.300000e+01 - 7.372260e+01 = -7.226000e-01 with absolute precision, but we get -7.226028e-01 due to floats having finite precision - with the error getting larger for larger values.
[... cut ...] culations the result is 0 and the test catches it an converts the value to `2*PI` as it should.

>  I’m guessing since GRBL is not a true Arduino sketch no debugging is possible?

The Arduino IDE does not support single stepping/debugging code - after all it is a hobbyist framework. And most Arduino boards do not have a debug header either?

FYI I made thes logs with my [STM32F7xx driver](https://github.com/grblHAL//STM32F7xx), it has a single precision FPU and supports double precision in code.

bz01t4x, 2023-02-18:
Thanks for the details!   So without thinking too hard about it, my takeaway is, depending on where the X, Y coordinates are, and depending on the radii and resultant radius vectors, the lack of precision shows up.  Seems like there should be a way of post-processing the G-code file to find and correct these possibilities?   Or, adding code to the GRBL source to trap negative values and bound the rt0 calc to zero?  I heard the GRBL code fills up the Uno uC - is there room to patch?

bz01t4x, 2023-02-18:
BTW, I looked at one of the STM32 boards - Nucleo-F756ZG.  Looks like a great $25 alternative.  Are the logs specific to the STM32 processor or does GRBL provide this?  I've been an embedded SW engineer for a long time (EV automotive) and have designed with a lot of different processors.  I'd consider moving to STM32 but would also like to see an 8 bit solution work and would be happy to try to develop one.  I'm a little surprised I haven't seen mention of this particular issue coming up before - things seem to have really picked up in GRBL use over the last 10 years.

terjeio, 2023-02-18:
> Seems like there should be a way of post-processing the G-code file to find and correct these possibilities?

This could be a viable solution, by transforming full circle arcs to two or more semicircles. But practical?

> Or, adding code to the GRBL source to trap negative values and bound the rt0 calc to zero?

Yes, that could be a solution - but what would a reasonable value be to test against? The precision loss [increases with larger values](https://upload.wikimedia.org/wikipedia/commons/thumb/1/18/IEEE754.svg/550px-IEEE754.svg.png)...

> I heard the GRBL code fills up the Uno uC - is there room to patch?

Perhaps, if stuck with a 8-bit MCU the Mega is a better option? Or switch to a patched 32-bit version? I am going to try double math in mine - at least this cannot have, IMO, any side-effects.

terjeio, 2023-02-18:
> BTW, I looked at one of the STM32 boards - Nucleo-F756ZG. Looks like a great $25 alternative.

[There are others](https://github.com/grblHAL/Controllers) that might be easier to set up...

>  Are the logs specific to the STM32 processor or does GRBL provide this?

No and no. I added some temporary code to output the logs.

> I'm a little surprised I haven't seen mention of this particular issue coming up before

I am as well, could be due to CAM post processors not outputting full circles in order to avoid controller issues?

terjeio, 2023-02-19:
> More like just simulating the code forcing restricted floating point library, catching violations and somehow adjusting best-fit X Y and/or G02 code to produce sub-optimal but at least approximately sized and placed circles.

Not something I would try. You would need an 100% accurate representation of the floating point code on the target MCU, access to the full 32bits of several float variables (offsets, initial position? etc). It would be far easier to do the suggested transform - and that would be accurate to boot.

> One of the things I wondered about was how UGS 'checks' things like this.

It cannot. If you are pondering why it manages to plot the circles correctly then that is likely due to using double precision variables internally (like I do in my sender).

> aren't typical hobby CNC/Laser cutters "smallish"

Not neccesarily - [some of these](https://www.grbl.org/projects) are quite large?

>  Looks like the easiest option is just to migrate to STM32. 

Moving to 32 bit is, IMO, the only realistic option. 8-bit Grbl has not been updated for four years now and there are several more capable 32-bit ports available. Time to move on...

>  Really appreciate the deep dive.

Working on the grblHAL port/rewrite since 2016 has given me a partial insight on how Grbl works - so I pretty much knew instantly where to look for where it could go wrong. And I want to fix "bugs" like this, inherited from the Grbl source, in grblHAL. I leave it to 8-bit/Arduino experts to fix it for Grbl as it may be a tad more complicated than my solution for grblHAL. In fact my solution may not solve the issue completely - there could be other issues lurking related to float precision. The paper I linked to above is relevant in that regard.

Finally, two of the processors currently supported by grblHAL has double precision FPUs, at some point in the future I might release a grblHAL core that switches all float math to double precision...

bz01t4x, 2023-02-20:
What do you think is the best supported 32bit board to get?  I'd feel better about getting involved knowing there was spare memory to put code in  : )

I saw you have made some CNC boards to supplement a few of the Nucleo models.

bz01t4x, 2023-02-23:
What do you think is the best supported 32bit board to get?  I'd feel better about getting involved knowing there was spare memory to put code in  : )

I saw you have made some CNC boards to supplement a few of the Nucleo models.

Also, still a question - I didn't alter the position or size of the stipples from the file I originally extracted from.  If the position and size (effecting precision) are what govern what is created versus skipped, I'm still not sure why they would be missing in one file and created in the other.  The only thing different is one was shorter (hundreds vs 9).

terjeio, 2023-02-23:
Sorry for answering late, your last comment got lost in my inbox...

> What do you think is the best supported 32bit board to get?

Hard to tell without knowing what "best supported" means for you. E.g. more pins means more plugins can be supported, networking usually means the WebUI can be used etc. So each board has its pros and cons. The Nucleo-F756ZG has a ton of pins but is an overkill for a 3-axis machine with no extra I/O requirements. For high speed laser engraving nothing beats a Teensy 4.1 (iMXRT1062) due to extremely low interrupt latency. If replacing an 8-bit Grbl controller that works well then the Pi Pico or a Pi Pico W (RP2040) is cheap, and more than capable. And so on...

> I'd feel better about getting involved knowing there was spare memory to put code in : )

All but the STM32F103CB has spare memory, as long as you stay away from that youll be fine.

> I saw you have made some CNC boards to supplement a few of the Nucleo models.

I have a few bare boards available for Nucleo-64 (no components mounted), you have to source the components and solder them yourself. Components are surface mounted and some are relatively fine pitch.

> I'm still not sure why they would be missing in one file and created in the other.

It is due to float math beeing imprecise in an unpredictable way as experienced by the user.  E.g. by simply altering the G54 offset different stipples where left out for the same gcode. This, for me, is the smoking gun.