grbl/grbl #1824: Homing - Origin defined at negative limits of axis

TobiasLibera, 2023-02-18:
Hi,
I have the problem, that after homing, GRBL's MPos is:
```
x = -499.000
y = -199.000
z = -1.000
```
(The z position is right)
\
\
My Max Travel Settings ($130, $131, $132) are the default value:
```
x = 500
y = 200
z = 200
```

My guess is, that's because I inverted some values for axis and homing direction, but I'm not 100% sure.
So I will add my GRBL configuration down here. Maybe someone knows a fix.
Much thanks.
\
\
***GRBL Settings***
```
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=5 (dir port invert mask:00000101)
$4=0 (step enable invert, bool)
$5=1 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=3 (homing dir invert mask:00000011)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=400.000 (x, step/mm)
$101=400.000 (y, step/mm)
$102=400.000 (z, step/mm)
$110=500.000 (x max rate, mm/min)
$111=500.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=500.000 (x max travel, mm)
$131=200.000 (y max travel, mm)
$132=200.000 (z max travel, mm)
```

jahnj0584, 2023-02-18:
If you home to the lower left corner, this is "right"

For instance, my machine homes at the top left corner and is at x-1200 y0

Keep in mind all these numbers are relative and don't need to mean
anything.

zachrockafellow, 2024-02-01:
I've also had this issue, and despite going through about a dozen articles referring to the same problem (each with their own rabbit-hole of a fix - none of which worked for me), I'm still lost.

I'm using LaserGrbl v5.7 with a 3018 CNC that has a woodpecker 3.4 board. It homes just fine. Axes are moving in the right direction, but the machine coordinates on my axes end up completely inverted. Any file I try to load sends the x-axis flying over towards the '0' position, which all the way on the right side of the machine. It then tries to move to whatever positive value is in my gcode. As soon as it tries to go past 0, it hits the wall and starts clicking violently. It does this with every axis, as they are all negative values.

After setting the axis direction with $3, I then setup the homing direction with $23. Then I tried inverting the coordinate values via $2. I'm not entirely sure that's how that works, tbh. It did nothing to my configuration whatsoever, regardless of what value I put in after $2.

I've been at this non-stop for about a week now, and I'm so lost. Not sure where to go for help, but hopefully someone here can chime in.

Here are my Grbl settings:
$$ < $0=10
$1=25
$2=7
$3=6
$4=0
$5=1
$6=0
$10=0
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=1
$24=25.000
$25=500.000
$26=250
$27=1.000
$30=1000
$31=0
$32=1
$100=800.000
$101=800.000
$102=800.000
$110=2250.000
$111=2250.000
$112=600.000
$120=30.000
$121=30.000
$122=30.000
$130=270.000
$131=180.000
$132=43.000

I'm wanting the X and Y axis to home to bottom left of the bed (so the spindle is close, and the end of the bed is far), and the Z
axis to home at the top of the gantry, set the coordinates to 0, then start printing my file.

MartinMajewski, 2025-01-28:
> HOMING_FORCE_SET_ORIGIN // uncommented this line to force origin to be set at homed location

That's it! Thank you so much!