grbl/grbl #1825: Idle|MPos:0.000,0.000,0.000|FS:,0>>: empty String

giulio83-cc, 2023-02-27:
hello, I state that I am a neophyte and I do not have a good knowledge of electronics. I built a foam cutter with two axes (X and Y) managed by three nema 17 stepper motors (2 motors for X axis and 1 motor for Y axis). I downloaded GRBL 1.1. H and uploaded the sketch to the Arduino uno. When I open Universal Gcode Sender, and connect the machine, it responds regularly to manual movement commands. But after a few minutes it starts to disconnect and reconnect continuously generating the following error:
**** Connected to COM5 @ 115200 baud ****
Grbl 1.1h ['$' fr help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$200
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26250
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$320
$100 = 5.697    (X-axis travel resolution, step/mm)
$101 = 4.717    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 500.00    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$10 = 200.000    (Status report options, mask)
$131 = 20.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:,0>>: empty String
<Idle|MPos:0.000,0.000,0.000|FS:0,0
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:0,>>: Index 1 out of bounds for length 1
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:00>>: Index 1 out of bounds for length 1
[Error] Error while processing response <<Idle|Mos:0.000,0.000,0.000|FS:00>>: Index 1 out of bounds for length 1
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:0,|Ov:100,100,100>>: Index 1 out of bounds for length 1
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:,0>>: empty String
[Error] Error while processing response <<Idle|MPo:0.000,0.000,0.000|FS:,0>>: empty String
Idle|MPos:0.000,0.000,0.00|FS:0,0|WCO:12.463,18.868,-1.000>
[Error] Error while processing response <<Idle|MPos:0.00,0.000,0.000|FS:0,>>: Index 1 out of bounds for length 1
Idle|MPos:0.000,0.00,0.000|FS:0,0>
Idle|MPos:0.000,0.00,0.000|FS:0,0>
[Error] Error while processing response <<Idl|MPos:0.000,0.000,0.000|FS:0,>>: Index 1 out of bounds for length 1
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:0,>>: Index 1 out of bounds for length 1
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:,0>>: empty String
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:,0|Ov:100,100,100>>: empty String
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:00>>: Index 1 out of bounds for length 1
[Error] Error while processing response <<Idle|MPos:0.000,0.000,0.000|FS:0,>>: Index 1 out of bounds for length 1
**** Connection closed ****
Pendant URL: http://192.168.1.78:8080/

Idle|MPos:0.000,0.000,0.000|FS:,0>>: empty String

Can anyone help me figure out what the problem is
I use s.o. Windows 10

livius2, 2023-03-08:
I have same issue, exact same errors

breiler, 2023-03-08:
The response data looks corrupted. Make sure you have a properly shielded USB cable. 
I have also seen this happening with poor quality Arduinos.

You should read this:
https://www.cnccookbook.com/cnc-electrical-noise-grounding-techniques-interference-filter/

rudemutant, 2023-03-23:
I'm no expert (I actually am), but I think there might be a timing problem somewhere? I'm not sure because I'm exhausted from trying to fix my mill. I reproduced the error. In my opinion, it looks like the text stream is having some rhythm issues. I'm running this on the mega328. I've used the precompiled hex file. I've compiled my own. I've tried different processors. I've tried the NANO. I've tried six different usb cables. I've tried OctoPrint, UGS, and pronterface. I've twisted all my wires. I've put chokes on EVERYTHING. I've shielded my mill until it looks like a crazy person hiding it's thoughts from the CIA. I've now unplugged my mill, taken the controller out, plugged it into a computer that is much more convenient (3rd host computer), had it run a simple program, ran it until it crashed (UGS is the most robust host, it didn't have a fit with the first error), and pasted the output:

Grbl 1.1h ['$' for help]
>>> $$
$1 = 25    (Step idle delay, milliseconds)
$0 = 10    (Step pulse time, microseconds)
$2 = 0    (Step pulse invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$3 = 1    (Step direction invert, mask)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 0    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$210
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$7 = 1.000   
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 1600.000    (X-axis travel resolution, step/mm)
$102 = 1600.000    (Z-axis travel resolution, step/mm)
$101 = 1600.000    (Y-axis travel resolution, step/mm)
$111 = 1000.000    (Y-axis maximum rate, mm/min)
$110 = 1000.000    (X-axis maximum rate, mm/min)
$112 = 1000.000    (Z-axis maximum rate, mm/min)
$120 = 500000    (X-axis acceleration, mm/sec^2)
$121 = 500.000    (Y-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$122500.000
$132 = 200.000    (Z-axis maximum travel, millimeters)
$131 = 00.000    (Y-axis maximum travel, millimeters)
>>> $G
ok
ok
[GC:G0 G54 G17 G21 90 G94 M5 M9 T0 F0S0]
Idle|WPos:0.000,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:0,0
Error while processing response <<Idle|WPos:0.000,0.000,0.000|FS:00|Ov:100,100,100>>: Index 1 out of bounds for length 1
Error while processing response <<Idle|Wos:0.000,0.000,0.000|FS:,0>>: empty String
Error while processing response <<Idle|WPos:0.000,0.000,0.000|FS:,0>>: empty String
Idle|WPos:0.000,0.00,0.000|FS:0,0|Ov100,100,100>
Error while processing response <<Idle|WPos:0.000,0.000,0.000|FS:00>>: Index 1 out of bounds for length 1
<Idle|WPos:0.000,0.000,0.000|FS:0,0
Idle|WPos:0.00,0.000,0.000|FS:0,0>
<Idle|WPos:0.000,0.000,0.000|FS:00
Error while processing response <<Idle|WPos:0.000,0.000,0.000|FS:00O:100,100,100>>: For input string: "00O:100"
Ile|WPos:0.000,0.000,.000|FS:0,0|Ov:100,100,100>
Idle|WPos:0.000,0.000,0.000|FS:0,0>
Idle|WPos:0.000,0.000,0.000|FS:0,0>
Error while processing response <<Idle|WPos:0.00,0.000,0.000|FS:0,>>: Index 1 out of bounds for length 1
Error while processing response <<Idle|WPo:0.000,0.000,0.000|FS:00>>: Index 1 out of bounds for length 1
Error while processing response <<Idle|WPos:0.000,0.000,0.000|FS:00>>: Index 1 out of bounds for length 1
Error while processing response <<Idle|WPos:0.0000.000,0.000|FS:00>>: Index 1 out of bounds for length 1
Error while processing response <<Idle|Wo:0.000,0.000,0.000|FS:0,>>: Index 1 out of bounds for length 1
Error while processing response <<Idle|WPos:0.000,0.000,0.000|F:,0>>: For input string: ",0"
Idle|WPos:0.000,0.000,0.000|FS:0,0>
Error while processing response <<Idle|Wos:0.000,0.000,0.000|FS:,>>: Index 0 out of bounds for length 0
Ile|WPos:0.000,0.000,.000|FS:0,0>
Error while processing response <<Idle|WPos:0.000,0.000,0.000|FS:00|Ov:100,100,100>>: Index 1 out of bounds for length 1
Idle|WPos:0.000,0.000,0.000|FS:0,0>
Error while processing response <<Idle|WPos:0.000,0.000,0.000|FS:0,|Ov:100,100,100>>: Index 1 out of bounds for length 1
Error while processing response <<Idle|WPos:0.000,0.000,0.000|FS:,0>>: empty String
Error while processing response <<Idle|WPos:0.000,0.000,0.000|F:,0>>: For input string: ",0"
[... 818 lines cut ...]
Error while processing response <<Run|Wos:16.788,100.810,0.000|FS:4000>>
Run|WPos:16.873,125.565,0.00|FS:400,0>
Error while processing response <<Run|WPos:16.985,158.646,0.000|FS:400,0WCO:0.000,0.000,0.000>>: For input string: "0WCO:0.000"
<Run|WPos:37.2531300,0.000|FS:400,0
<Run|WPos:3.000,153.494,0.000|FS:400,0
Error while processing response <<Run|WPos:73.000,105.446,0.000|FS:4000>>
<Run|WPos:73.000,98.564,0.000|FS:400,0
<Run|WPos:73.00,67.034,0.000|FS:400,0
<Run|WPs73.000,35.473,0.000|FS:400
Error while processing response <<Run|WPos:51.754,17.000,0.000|FS:4000Ov:100,100,100>>: For input string: "4000Ov:100"
<Run|WPs:41.973,17.000,0.000|FS:4000
Error while processing response <<Run|WPos:17.013,20.571,0.000|FS:40,>>
Error while processing response <<Run|WPos:17.050,31.509,0.000|FS:400,>>
Error while processing response <<Run|WPos:17.093,43.851,0.000|FS:400,>>
[... cut ...]
Error while processing response <<Run|WPos:17.229,83.773,0.000|FS:40,>>
Error while processing response <<Run|WPos:17.277,97.497,0.000|F:0,>>: For input string: "0,"
Run|WPos:17.286,100.221,0.000|FS:400,0>
Error while processing response <<Run|WPos:7.389,130.326,0.000|FS:4000>>
<Run|WPos:17.399,133.087,0.000|FS:400,0
Idle|WPos:17.500,162.500,0.000|FS:0,0|Ov:100,100,100>
Idle|WPos:17.500,162.500,0.000|FS:0,0>
Error while processing response <<Idle|WPos:17.500,162.500,0.000|FS:0,|WCO:0.000,0.000,0.000>>
Idle|WPos:17.500,162.5000.000|FS:0,0>

rudemutant, 2023-03-23:
Breiler, It won't even run on my fancy arduino mega. It just sits there and does nothing. You do realize that many of these controllers in your user base are "cheap chinesium arduino-clone"s... You've generally identified the problem, and specifically blamed the users.

breiler, 2023-03-23:
> I've tried my best to be a strong supporter of the open source community. In order for us to keep our projects working, we need feedback on errors. I provided feedback.

So your idea with being a strong supporter of open source is to deliver your feedback to the community with an attitude?

I answered a stale issue that gets asked pretty often, the best way I could, with the best knowlege that I have/had. I have no way of reproducing this error as I have thrown away all my bad Arduinos in favor for the ones that work.

You proved my assumptions about EMI as the culprit to be wrong and I am glad that I learned something new today. I might consider implementing some kind of test for this in UGS (which I feel a bit responsible for).

As you have tried several different senders with the same result we can rule out that it is a problem with how the senders are communicating with GRBL. You mentioned that you have several Arduinos with the same problem. I am a bit curious about what USB serial chip they are using.

rudemutant, 2023-03-23:
57600 works fine. I would agree with you on the EMI thing, but 500000 actually runs better than 115200 . 500000 is still not great, but whatever timing issues I have are gone with 56K, and that is fine. UGS was showing it out of order in the above. Arduino is showing the correct response to $$. Is it your buffer? Is it my crap UART chip?

Thanks for your service and patience
XOXO 
-Rudemutant

celalmutlu, 2023-03-23:
Grbl 1.1 ['$' for help]
[MSG:Check limits]
[MSG:'$H'|'$X' to unlock]
[MSG:Caution: Unlocked] 
What can i do?

AlexHolden, 2023-03-24:
It's most likely something to do with the USB to serial interface chip on the Arduino. The CH340 used on most Chinese clone Arduinos is known to be unreliable and unfixable. I stopped using a clone Nano with a CH340 on it because I was getting random corruption. The Atmega16u2 used on official Arduinos (used to? maybe still does?) ship with buggy firmware that works fairly reliably up to 115200 baud but randomly corrupts data at higher throughput. About six years ago I posted a patched version of the 16u2 firmware that has been 100% reliable for me at 250000 baud:
https://github.com/grbl/grbl/issues/845#issuecomment-160190898

Note: I'm not a grbl developer; I just use it in production on my CNC milling machine. I don't think anyone is actively developing this old version of grbl any more. https://github.com/gnea/grbl is the current version for 8 bit AVR Arduinos, and most active development  these days seems to be for 32 bit boards over at https://github.com/grblHAL

rudemutant, 2023-03-24:
> Grbl 1.1 ['$' for help] [MSG:Check limits] [MSG:'$H'|'$X' to unlock] [MSG:Caution: Unlocked] What can i do?

Celalmutlu, your question should be put under a different thread, however: I'm 90% sure you are the target of a passive aggressive message telling you to home your machine. There is an option in the firmware (that I either turned off, or was off by default), and it will lock itself until it sees a $H... and this is probably the wrong project as I just noticed above. Sorry Breiler