grbl/grbl #1833: problems setting up my arduino uno to ugs

Simonloff, 2023-04-26:
When i try to configure my arduino uno to ugs with the setup wizard it tells me "this set up wizard does not know how to configure this type of controller. it needs to be configured manually" how do i configure it manually or how can i arrange the problem of the setup wizard?

breiler, 2023-04-29:
What version of GRBL and UGS are you using?

Simonloff, 2023-05-02:
no worries problem solved :)

abdx79, 2023-08-07:
> 

I'm facing the same issue; how did you solve that.

Yashk09, 2023-10-05:
I have same problem 
Please tell me something about that

breiler, 2023-10-05:
@Yashk09 please create a new issue or discussion here: https://github.com/winder/Universal-G-Code-Sender

Please provide the what controller you have and which version of UGS you are running.