grbl/grbl #1835: Erratic motor motion after change from estlcam to grbl

Keshly, 2023-05-08:
Hello, 

I'm really getting desperate. I have the following setup, which works perfectly with Ectlncam. With Grbl I just can't get it to work.

Arduino Mega

Drivers: DM860 (1600=8)
Motors: Nema34 (1.8=200)

estlcam Settings:
(works excellently and extremely quietly)

Steps per revolution: 1600
Distance per revolution 24,13
Maximum feedrate: 8000mm/min
Inertia: 85%
Step pulse length: 5µs
Step pause: 0

grbl/(*-Mega-5X) settings: *with normal grbl, there is the same problem as with the fork 5X
(erratic motor motion, stuttering, losing Steps, extreme loudness)

$0=5			Step pulse, microseconds	DEFAULT_STEP_PULSE_MICROSECONDS
$1=254			Step idle delay, milliseconds	DEFAULT_STEPPER_IDLE_LOCK_TIME
$2=0			Step port invert, mask	DEFAULT_STEPPING_INVERT_MASK
$3=0			Direction port invert, mask	DEFAULT_DIRECTION_INVERT_MASK
$4=0			Step enable invert, boolean	DEFAULT_INVERT_ST_ENABLE
$5=0			Limit pins invert, boolean	DEFAULT_INVERT_LIMIT_PINS
$6=0			Probe pin invert, boolean	DEFAULT_INVERT_PROBE_PIN
$10=2			Status report, mask	DEFAULT_STATUS_REPORT_MASK
$11=0.02 		Junction deviation, mm	DEFAULT_JUNCTION_DEVIATION
$12=0.002 		Arc tolerance, mm	DEFAULT_ARC_TOLERANCE
$13=0			Report inches, boolean	DEFAULT_REPORT_INCHES
$20=1			Soft limits, boolean	DEFAULT_SOFT_LIMIT_ENABLE
$21=0			Hard limits, boolean	DEFAULT_HARD_LIMIT_ENABLE
$22=1			Homing cycle, boolean	DEFAULT_HOMING_ENABLE
$23=0			Homing dir invert, mask	DEFAULT_HOMING_DIR_MASK
$24=25.000		Homing feed, mm/min	DEFAULT_HOMING_FEED_RATE
$25=250			Homing seek, mm/min	DEFAULT_HOMING_SEEK_RATE
$26=250			Homing debounce, milliseconds	DEFAULT_HOMING_DEBOUNCE_DELAY
$27=5			Homing pull-off, mm	DEFAULT_HOMING_PULLOFF
$30=12000		Max spindle speed, RPM	DEFAULT_SPINDLE_RPM_MAX
$31=550			Min spindle speed, RPM	DEFAULT_SPINDLE_RPM_MIN
$32=0			Laser mode, boolean	DEFAULT_LASER_MODE
$100=66.308		Axis 1 (X) steps/unit (mm or deg)	DEFAULT_AXIS1_STEPS_PER_UNIT
$101=66.308		Axis 2 (Y) steps/unit	DEFAULT_AXIS2_STEPS_PER_UNIT
$102=66.308		Axis 3 (Z) steps/unit	DEFAULT_AXIS3_STEPS_PER_UNIT
$103=8.889		Axis 4 (A) steps/unit	DEFAULT_AXIS4_STEPS_PER_UNIT
$110=8000		Axis 1 (X) Max rate, unit/min	DEFAULT_AXIS1_MAX_RATE
$111=8000		Axis 2 (Y) Max rate, unit/min	DEFAULT_AXIS2_MAX_RATE
$112=8000		Axis 3 (Z) Max rate, unit/min	DEFAULT_AXIS3_MAX_RATE
$113=4000		Axis 4 (A) Max rate, unit/min	DEFAULT_AXIS4_MAX_RATE
$120=85			Axis 1 (X) Acceleration, unit/sec^2	DEFAULT_AXIS1_ACCELERATION
$121=85			Axis 2 (Y) Acceleration, unit/sec^2	DEFAULT_AXIS2_ACCELERATION
$122=85			Axis 3 (Z) Acceleration, unit/sec^2	DEFAULT_AXIS3_ACCELERATION
$123=85			Axis 4 (A) Acceleration, unit/sec^2	DEFAULT_AXIS4_ACCELERATION
$130=1723.500		Axis 1 (X) Max travel, unit	DEFAULT_AXIS1_MAX_TRAVEL
$131=1825.500		Axis 2 (Y) Max travel, unit	DEFAULT_AXIS2_MAX_TRAVEL
$132=200.000		Axis 3 (Z) Max travel, unit	DEFAULT_AXIS3_MAX_TRAVEL
$133=360.000		Axis 4 (A) Max travel, unit	DEFAULT_AXIS4_MAX_TRAVEL

Note: all axes move in the right direction and can be controlled separately.
I have already tried unsuccessfully by changing any number of parameters in config.h, defaults.h. However, I can't seem to find the right parameter. (Of course I restored the original settings after each change)

The strange thing is that esctlcam doesn't let you adjust much. As soon as I program the control via esctlcam (without any hardware changes), everything works fine again... But since I can't operate a 4th axis here, I would like to use GRBL.

I would be very grateful to the solution provider, since I'm already spending a few days of my vacation on this problem :(

Thanks in advance.

Greetings
Chris

breiler, 2023-05-09:
The max rate seems a bit high to me. It depends on what type of machine you have of course but  8000 mm/min is a lot. Are you sure that you could run at those speeds with EstlCAM?

I am thinking that a simple test would be to try and run one of these commands at the time and see where the motors stalls:
```
G1 G91 G21 X200 F1000
G1 G91 G21 X-200 F2000
G1 G91 G21 X200 F3000
G1 G91 G21 X-200 F4000
G1 G91 G21 X200 F5000
G1 G91 G21 X-200 F6000
G1 G91 G21 X200 F7000
G1 G91 G21 X-200 F8000
```

Keshly, 2023-05-09:
@breiler 
Thanks for your reply. in EstlCAM 120000 works with my setup, only then do vibrations start. in GRBL I have vibration and noise even at speeds 500-2000... if I increase #define ACCELERATION_TICKS_PER_SECOND 100 then it will be a little less vibration
disable "Comment" // #define ADAPTIVE_MULTI_AXIS_STEP_SMOOTHING also reduced the vibration, but I don't think that's a good solution...