grbl/grbl #1843: I can't find full video of how to make connections

lidershinobi, 2023-06-29:
I need someone to explain to me the connection of the drill motor that is activated with the board and that uses an inductive sensor for homing, there is no complete video that explains how to make all those connections and the configuration in arduino

Sualc, 2023-07-01:
Hello, perhaps this https://www.youtube.com/watch?v=hhlSpaA8an4 may help you.
Probably not if you don't use one variable speed driver like the video.

Because you said "activated" I suppose you don't think in speed control. In this case you will need a driver to give the power for your spindle. An output from arduino board have very low power capacity. You can use  something like a relay controlled by one transistor and this transistor connected to the pin 11 like show in the video. In your gcode you can set the speed ("Sxx..x" command) S0 will be Drill OFF and some bigger value (S20000 i.e.) will activate the transistor => Drill ON.
Good luck.

lidershinobi, 2023-07-01:
what you need to know is where the pwm signal is connected in the CNC Shield V4 controller kit with arduino nano