grbl/grbl #1847: Upload Issues

perryismyname, 2023-08-09:
I have had nonstop issues with GRBL and UGS. I have been trying for about three weeks now and I am getting fed up at this point. I don't know if I am doing something wrong but I have started from scratch 15+ times and have followed every tutorial video I could find on youtube. I am the closest I have been to getting it running now so I don't want to restart again.

I am running arduino 1.8.19 as arduino 2.0 can't select the grbl folder (it just opens into the subfolder)
Using an elegoo uno R3, cnc shield v3, A4988s, nema 17s, and standard everything else

The root problem I think comes with uploading GRBL. I have redownloaded it several times all of which give me the same error while uploading. I have attached the code below. I have also attempted to redownload arduino programming several times to no avail. Somehow when I completely uninstall it the libraries remain in the program after I install a new program, even if I have everything set to a separate file. Maybe I am not actually uninstalling it then. Ultimately I got the whole thing to run in super basic forms. It is a 2 axis (x axis cloned) laser cutter so nothing is too crazy about it. I was only able to get it to do basic jogs, I can not get it to home or pass UGS setup wizard (due to homing issue). The issue is not false flagging of limit switches either.

Any help would be appreciated. I am so peeved at this point.

`In file included from C:\Users\johns\Documents\Arduino\libraries\grbl/grbl.h:43:0,
                 from C:\Users\johns\Documents\Arduino\libraries\updated3\examples\grblUpload\grblUpload.ino:27:
C:\Users\johns\Documents\Arduino\libraries\grbl/nuts_bolts.h:56:0: warning: "max" redefined
 #define max(a,b) (((a) > (b)) ? (a) : (b))
 
In file included from sketch\grblUpload.ino.cpp:1:0:
C:\Users\johns\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/Arduino.h:93:0: note: this is the location of the previous definition
 #define max(a,b) ((a)>(b)?(a):(b))
 
In file included from C:\Users\johns\Documents\Arduino\libraries\grbl/grbl.h:43:0,
                 from C:\Users\johns\Documents\Arduino\libraries\updated3\examples\grblUpload\grblUpload.ino:27:
C:\Users\johns\Documents\Arduino\libraries\grbl/nuts_bolts.h:57:0: warning: "min" redefined
 #define min(a,b) (((a) < (b)) ? (a) : (b))
 
In file included from sketch\grblUpload.ino.cpp:1:0:
C:\Users\johns\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/Arduino.h:92:0: note: this is the location of the previous definition
 #define min(a,b) ((a)<(b)?(a):(b))
 
In file included from C:\Users\johns\Documents\Arduino\libraries\grbl/grbl.h:43:0,
                 from C:\Users\johns\Documents\Arduino\libraries\updated3\examples\grblUpload\grblUpload.ino:27:
C:\Users\johns\Documents\Arduino\libraries\grbl/nuts_bolts.h:61:0: warning: "bit" redefined
 #define bit(n) (1 << n)
 
In file included from sketch\grblUpload.ino.cpp:1:0:
C:\Users\johns\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/Arduino.h:124:0: note: this is the location of the previous definition
 #define bit(b) (1UL << (b))
 
Sketch uses 29762 bytes (92%) of program storage space. Maximum is 32256 bytes.
Global variables use 1633 bytes (79%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.
`

breiler, 2023-08-09:
The messages you get when compiling are just warnings, so it should be ok.

You are not describing why you think there is a problem other than homing. If I had to guess, a common problem when trying to home using a two axis system is that GRBL is configured to use three axes by default. Check the file `C:\Users\johns\Documents\Arduino\libraries\grbl\config.h` on this line https://github.com/grbl/grbl/blob/master/grbl/config.h#L75 

Change the homing sequence by replacing the lines:
```
#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
```

to this:
```
#define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))                
```

perryismyname, 2023-08-09:
> The messages you get when compiling are just warnings, so it should be ok.
>
> You are not describing why you think there is a problem other than homing. If I had to guess, a common problem when trying [...]