grbl/grbl #1850: GRBL feed control with potentiometer

timeO25, 2023-09-21:
Hello,
is it possible to control the feed with a potentiometer on grbl?

Best reguards
timeO25

breiler, 2023-09-24:
No, GRBL has no such feature. Instead you'd need a separate PWM spindle motor controller for this.

timeO25, 2023-09-24:
Thanks for help I'll try it.

timeO25, 2023-10-10:
Do you plan to allow the feedrate to be adjust in an upcomming grbl?

What are these following commands how can I use them on serial connection like Arduino IDE?

The wiki says the following:
0x90 : Set 100% of programmed rate.
0x91 : Increase 10%
0x92 : Decrease 10%

breiler, 2023-10-10:
> Do you plan to allow the feedrate to be adjust in an upcomming grbl?

There is no more development being done on either the classic GRBL 0.9 or [Gnea/GRBL](https://github.com/gnea/grbl). Have a look at newer projects such as [grblHAL](https://github.com/grblHAL) or [FluidNC](https://github.com/bdring/FluidNC).

The commands that you refer to are real time commands and is sent by the gcode sender. Most senders support these such as UGS, CNCJS, bCNC or Candle.