grbl/grbl #1852: X axis losses position only when another axis is moving

AugustoQOJ, 2023-10-14:
I am making a CNC machine and managed to get it ruining,  so i started making some parts when i notice that some of them were having errors on the measure, now for my purpose it was not a big deal at first, but then it would start to get worst over time, and for some programs it didn't  even finished the thing  before catastrophically losing position. 

At first i thought it was mechanical because it was weirdly consistent to be the wiring or EMI (even with my wiring being garbage and heaving no protection against EMI). So i looked and i did find that the couplers were going loose a LOT, but this only solve the problem partially. So i took the motor of the thing and ran de program that made the most error. And lo and behold it got out of place. (The zero i marked was not the zero the motor stopped on). I have gone to the possible causes.

1. I had to switch a motor because the other one was too weak to do the job, but it was WAY stronger and it could be making the font unable to power anything. But the error continued even after disconnecting it.
2. EMI was the next obvious one, so i tried taking all wires out of the Arduino to see if the error diminished at least, it did not.
3. The driver or the motor may be faulty, so i switched those and still got the error.

So know i can only see 3 possibilities, either the Arduino is to blame, or the shield, or grbl itself. There is also de 4th (and most likely) option that i made a mistake and just don't know all that is needed to solve this (it is my first time doing something like this).
Although it may be the computer i am using to send the GCODE since it is very old.

findings:
It is not loosing steps, even when it runs in the negative direction it still losses in the positive side, meaning it is going further on the positive than it should be (when i set the direction to be inverted the error happens the new positive direction).

The other axis are loosing position, but to a lesser extent than X

The error only occurs when 2 or more axis move in alternating turns. If X and Y run at the same time, or only X, it doesn't lose position. 
But if X runs, then stops, and then Y and/or Z runs then there is an error (and it takes a lot for the error to be noticeable by the naked eye).

Is is not interference that is causing the problem (probably), not only because i ran the X without any of the other axis connected to the Arduino, but also because when i run a program that makes X on the same place (always G0 X0) while alternating the Y axis (or any axis for that matter) it didn't made an error.

I already re uploaded grbl to the Arduino, even resetting the settings to default, to see if it got better, it didn't.

Could someone please help me, i am out of ideas on how to catch this zebra.  

Info:

GRBL version: 0.9j

Sending Gcode using UGS on a 22.0 version ubuntu on a old desktop i had lying around, (It has a tendency to crash).

GRBL config: the default ones but the steps per mm of the X axis is 320 and its speed and accel are in half. i would put the whole thing here but i currently don't have access to it. if needed i will post here as soon as i can.

Drivers: X: DM542
             Y: HBS86H with a encoder
             Z: Another DM542

Any help would be very appreciated. And sorry for the long text. 

P.S: sorry about misspellings, English isn't my first language.

jahnj0584, 2023-10-14:
Sounds like not enough amperage. What's the power supply, and is there one unique one for each axis?

AugustoQOJ, 2023-10-15:
I don't remember the power supply specifically but it is powering like everything. But then why it would gain steps and not loosing them?

langwadt, 2023-10-15:
just a random thought, DM542 needs the direction signal valid quite a long time (+5us) before the step pulses, is your grbl build with  STEP_PULSE_DELAY (in config.h) defined?

AugustoQOJ, 2023-10-16:
i don't know, i will take a look.

AugustoQOJ, 2023-10-16:
> just a random thought, DM542 needs the direction signal valid quite a long time (+5us) before the step pulses, is your grbl build with STEP_PULSE_DELAY (in config.h) defined?

thanks for the advice, it wasn't declared. But the problem persisted even when i set the value to 300.