grbl/grbl #1856: The spindle/laser output stuck

realbukaPL, 2023-12-04:
I have a CNC shield on arduino UNO controlled by Universal Gcode Platform v. 2.0. Machine works fine, but I cannot get any output from M3 nor M5 commands. I tried it in laser mode, tried switching $30 values, try different output pins. I check output with a meter and always get 4.5V on pin D12, that goes DOWN to 4.44V when there should be output (spindle) from a program. 
I'm on it for hours now and I can't find anything wrong with connections in any combination. 

Is it some weird PWM setting? And why is it dropping this couple of milivolts? I just want a normal of and on 0-5v signal to use with my printing head. 

I don't have any alarms or warnings going off

breiler, 2023-12-05:
Post the output from `$I`. Exactly what shield did you buy (provide a link to it).

realbukaPL, 2023-12-05:
Thank you for your inteterest

The board is:
[image]

My grlb version is 1.1h.20190830, opt v, 15,128

> Post the output from `$I`. Exactly what shield did you buy (provide a link to it).

Link is in polish and will not be of use, but I may add it is clearly the cheap version and does not look original

langwadt, 2023-12-05:
this is the old repo, you should be looking here: https://github.com/gnea/grbl

breiler, 2023-12-05:
@langwadt is right, the version 1.1 of GRBL is using this repository: https://github.com/gnea/grbl

But the shield that you have bought might still use the old GRBL <0.9 pin mapping.

This is the default pin mapping of GRBL 1.1:
https://github.com/gnea/grbl/wiki/Connecting-Grbl#grbls-pins

Since you didn't provide the link to the product, I will have to assume it is based on the protoneer design which expects a different pin mapping: https://blog.protoneer.co.nz/arduino-cnc-shield/#BoardLayout

realbukaPL, 2023-12-05:
@breiler I understand, this would explain a lot. Thank you for your assistance.

realbukaPL, 2023-12-05:
It seems the issue is mismatched hardware