grbl/grbl #1859: grbl 1.1h spindle with relay shield ( Buzzing Relay)

NyuszyFul, 2023-12-17:
I have a problem activating the relay for the spindle! I tried connecting it to the Z+ pin on Shield v3, but the relay keeps buzzing continuously. Even when I attempt to trigger the function from the UGS software console, nothing happens. For instance, if I type "m03 s10000," the buzzing intensifies, but the spindle still doesn't turn. If I revert to the console command "m03 s0," the buzzing decreases, but it persists.

Could this be interference? When I connect the relay to any +5V pin, it engages normally, and the spindle turns. Can you help me figure out what to do?

I'll attach some media files for you to review.

https://github.com/grbl/grbl/assets/104527898/63434f48-847c-4c40-9f58-31df20ee2c86

NyuszyFul, 2023-12-17:
Uploading WhatsApp Video 2023-12-17 at 20.23.36.mp4…

breiler, 2023-12-17:
You haven't given any details on your configuration, so I will have to assume that you are using GRBL 0.9 with PWM enabled. Driving a relay with PWM will not work.

If you want to use a relay try GRBL 0.9 without PWM and use the spindle enable (D12) instead.

NyuszyFul, 2023-12-17:
> D12

i will try 

i still try by my self some setups and with the 1.1h i can turn  it **ON** with _m4 s1000_ and **OFF** with _m3 s0_ if i put the cable on SpnDir