grbl/grbl #1867: homing issue on my cnc 3018 pro using ugs

plceleste, 2024-01-23:


plceleste, 2024-01-23:
$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
$3 = 6    (dir port invert mask:00000110)
$4 = 0    (step enable invert, bool)
$5 = 0    (limit pins invert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 2    (status report mask:00000010)
$11 = 0.010    (junction deviation, mm)
$12 = 0.002    (arc tolerance, mm)
$13 = 0    (report inches, bool)
$20 = 0    (soft limits, bool)
$21 = 1    (hard limits, bool)
$22 = 1    (homing cycle, bool)
$23 = 1    (homing dir invert mask:00000001)
$24 = 25.000    (homing feed, mm/min)
$25 = 500.000    (homing seek, mm/min)
$26 = 250    (homing debounce, msec)
$27 = 1.500    (homing pull-off, mm)
$30 = 10000    (rpm max)
$31 = 0    (rpm min)
$32 = 0    (laser)
$43 = 0    (sp en invert)
$44 = 0    (sp mode)
$45 = 5    (fg pulse/rv )
$46 = 0    (ssm mode)
$51 = 8192    (id)
$52 = 1000066    (ver)
$53 = 578618160    (sn)
$54 = 1561217489    (date)
$55 = 1000380    (pver)
$56 = 4224403738    (k)
$57 = 1904353242    (f)
$58 = 4607050    (info0b)
$59 = 0    (info4b)
$60 = 0    (info8b)
$61 = 0    (info12b)
$100 = 800.000    (x:stp/mm)
$101 = 800.000    (y:stp/mm)
$102 = 800.000    (z:stp/mm)
$103 = 50000.000    (a:stp/mm)
$104 = 50000.000    (b:stp/mm)
$105 = 3200.000    (c:stp/mm)
$110 = 2000.000    (x:max rate, mm/min)
$111 = 2000.000    (y:max rate, mm/min)
$112 = 500.000    (z:max rate, mm/min)
$113 = 500.000    (a:max rate, mm/min)
$114 = 500.000    (b:max rate, mm/min)
$115 = 500.000    (c:max rate, mm/min)
$120 = 3000.000    (x:accel, mm/s^2)
$121 = 3000.000    (y:accel, mm/s^2)
$122 = 5000.000    (z:accel, mm/s^2)
$123 = 10.000    (a:accel, mm/s^2)
$124 = 10.000    (b:accel, mm/s^2)
$125 = 10.000    (c:accel, mm/s^2)
$130 = 270.000    (x:max travel mm)
$131 = 170.000    (y:max travel mm)
$132 = 15.000    (z:max travel mm)
$133 = 200.000    (a:max travel mm)
$134 = 200.000    (b:max travel mm)
$135 = 200.000    (c:max travel mm)

this is my firmware setting

breiler, 2024-01-23:
Previous discussion can be found here: https://github.com/grbl/grbl/issues/1696

These settings are not "standard GRBL" so Benbox is running their own fork. The settings which I thought were corrupt are their own.

```
$43 = 0 (sp en invert)
$44 = 0 (sp mode)
$45 = 5 (fg pulse/rv )
$46 = 0 (ssm mode)
$51 = 8192 (id)
$52 = 1000066 (ver)
$53 = 578618160 (sn)
$54 = 1561217489 (date)
$55 = 1000380 (pver)
$56 = 4224403738 (k)
$57 = 1904353242 (f)
$58 = 4607050 (info0b)
$59 = 0 (info4b)
$60 = 0 (info8b)
$61 = 0 (info12b)

$103 = 50000.000 (a:stp/mm)
$104 = 50000.000 (b:stp/mm)
$105 = 3200.000 (c:stp/mm)

$113 = 500.000 (a:max rate, mm/min)
$114 = 500.000 (b:max rate, mm/min)
$115 = 500.000 (c:max rate, mm/min)

$123 = 10.000 (a:accel, mm/s^2)
$124 = 10.000 (b:accel, mm/s^2)
$125 = 10.000 (c:accel, mm/s^2)

$133 = 200.000 (a:max travel mm)
$134 = 200.000 (b:max travel mm)
$135 = 200.000 (c:max travel mm)
```

I can't access their website (www.benbox.cn) as it is timing out - so no documentation or specs for the controller board.  According to the aliexpress page it is running om a Atmel 328p: https://vi.aliexpress.com/item/4001102703498.html?gatewayAdapt=glo2vnm

Thinking about your statement that when you trigger a limit switch, all pins are reported as triggered in UGS. I wonder if they have consolidated the inputs for all limit pins to be able to add an extra axis? 🤷🏻‍♂️ 

The GRBL license requires them to provide their modified source code which could shed some light on the problem. But since that is not available either makes things really difficult.

Sorry I can't help you, but maybe someone else can.

plceleste, 2024-01-24:
what do you mean by not standard?

breiler, 2024-01-24:
> what do you mean by not standard?

Benbox took the GRBL source code and added their own stuff to it.