grbl/grbl #1872: I encountered a problem with $$ command

woaeo, 2024-03-15:
It starts when servo on my plotter begin jamming time to time, so i deciede to replace it, i didnt have exact model, and put another with the same dimensions, i thought that wouldn't be an issue, but since than my plotter does not work correctly, even after i put back the old servo.
After every boot, when $$ command happens, it give me this:

[image]

full text:

```
$0=10 (step pulse, usec)
$1=5 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.000 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=2 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=250.000 (x, step/mm)
$101=250.000 (y, step/mm)
$102=250.000 (z, step/mm)
$110=20000.000 (x max rate, mm/min)
$111=20000.000 (y max rate, mm/min)
$112=500.000 (z max rate, mm/min)
$120=200.000 (x accel, mm/sec^2)
$121=200.000 (y accel, mm/sec^2)
$122=-2147483.648 (x accel, mm/sec^2)
$171=0.000 (x accel, mm/sec^2)
$172=0.000 (x accel, mm/sec^2)
$173=0.000 (x accel, mm/sec^2)
$174=0.000 (x accel, mm/sec^2)
$175=0.000 (x accel, mm/sec^2)
$176=0.000 (x accel, mm/sec^2)
$177=0.000 (x accel, mm/sec^2)
$178=0.000 (x accel, mm/sec^2)
$179=0.000 (x accel, mm/sec^2)
$180=0.000 (x accel, mm/sec^2)
$181=0.000 (x accel, mm/sec^2)
$182=0.000 (x accel, mm/sec^2)
$183=0.000 (x accel, mm/sec^2)
$184=0.000 (x accel, mm/sec^2)
$185=0.000 (x accel, mm/sec^2)
$186=0.000 (x accel, mm/sec^2)
$187=0.000 (x accel, mm/sec^2)
$188=0.000 (x accel, mm/sec^2)
$189=0.000 (x accel, mm/sec^2)
$190=0.000 (x accel, mm/sec^2)
$191=0.000 (x accel, mm/sec^2)
$192=0.000 (x accel, mm/sec^2)
$193=0.000 (x accel, mm/sec^2)
$194=0.000 (x accel, mm/sec^2)
$195=0.000 (x accel, mm/sec^2)
$196=0.000 (x accel, mm/sec^2)
$197=0.000 (x accel, mm/sec^2)
$198=0.000 (x accel, mm/sec^2)
$199=0.000 (x accel, mm/sec^2)
$200=0.000 (x accel, mm/sec^2)
$201=0.000 (x accel, mm/sec^2)
$202=0.000 (x accel, mm/sec^2)
$203=0.000 (x accel, mm/sec^2)
$204=0.000 (x accel, mm/sec^2)
[... 368 lines cut ...]
$105=0.000 (x accel, mm/sec^2)
$106=0.000 (x accel, mm/sec^2)
$107=0.000 (x accel, mm/sec^2)
$108=0.000 (x accel, mm/sec^2)
$109=0.000 (x accel, mm/sec^2)
$110=0.000 (x accel, mm/sec^2)
$111=0.000 (x accel, mm/sec^2)
$112=0.000 (x accel, mm/sec^2)
$113=0.000 (x accel, mm/sec^2)
$114=0.000 (x accel, mm/sec^2)
$115=0.000 (x accel, mm/sec^2)
$116=0.000 (x accel, mm/sec^2)
$117=0.000 (x accel, mm/sec^2)
$118=0.000 (x accel, mm/sec^2)
$119=0.000 (x accel, mm/sec^2)
$120=0.000 (x accel, mm/sec^2)
$121=0.000 (y accel, mm/sec^2)
$122=0.000 (z accel, mm/sec^2)
$130=-0.000 (x max travel, mm)
$131=-0.000 (y max travel, mm)
$132=-0.000 (z max travel, mm)
```

and i have no idea how to fix it, i tried flash it and reflas it many times, with different versions of grbl, and it still do the same.
I use Arduino Uno with v3 CNC Shield

woaeo, 2024-03-15:
p.s it really do accelerate slow, even when i change it

breiler, 2024-03-15:
>  i tried flash it and reflas it many times, with different versions of grbl, 

This is probably the issue. The eeprom memory has has different layouts between different versions which has probably made the settings corrupt. I think that you need to reset all settings: https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#rst-rst-and-rst--restore-grbl-settings-and-data-to-defaults

woaeo, 2024-03-15:
> I think that you need to reset all settings

Thanks, that work, but now i cant enable Corexy, even when it stated in config.h

woaeo, 2024-03-15:
> I think that you need to reset all settings:

THANKS alot