grbl/grbl #1874: GRBL homing problem when using the homing cycle in G-code after G1/G0 commands

Turner66, 2024-04-05:
Hello,

I'm encountering an issue with GRBL where homing commands within G-code aren't functioning as expected after movement commands. Specifically, I'm trying to initiate a homing cycle at specific points within my G-code sequence. 
For instance:

$H
G1 X100 F1000
G1 Y0
$H
G1 Z10 F100
G1 Y30

However, only the first homing command seems to work reliably. 
Once a homing command is issued after a G1/G0 movement command, subsequent homing commands do not function correctly. 

Interestingly, this works with Marlin firmware.

Has anyone encountered a similar issue with GRBL, and if so, is there a workaround or solution available? Any insights or suggestions would be greatly appreciated.

Kind regards
Tim

breiler, 2024-04-05:
`$H` is not a gcode command and should not be included in a gcode program.

Are you really using `$H` on Marlin or are you using their interpretation/implementation of `G28` (https://marlinfw.org/docs/gcode/G028.html)? 

Read about G28 in GRBL here: https://github.com/grbl/grbl/wiki/Frequently-Asked-Questions#why-are-some-of-grbls-g-codes-a-little-different-as-on-some-other-cnc-machines and here https://www.linuxcnc.org/docs/2.5/html/gcode/gcode.html#sec:G28-G28_1.

As far as I know you can't do homing as a part of gcode program in GRBL, this must be done through the sender.

Turner66, 2024-04-05:
Thanks for the quick reply. yes, in the marlin example I used g28 instead of $H. In marlin, as you wrote, g28 is the command for the homing cycle. unfortunately this is not the case in Grbl. maybe there is the possibility to add a G command for homing (like in Marlin).