grbl/grbl #1875: How do I use variable spindle speed when Z axis limit switch is in use?

Jandre3d, 2024-04-11:
Hey all

I recently finished my cnc, but I want to connect my spindle to the setup now (im turning it on manually now)
I have a relay, but i dont know how to connect since i disabled it to use a Z axis limit switch
Im using grbl 1.1h, I disabled variable spindle speed in the code

breiler, 2024-04-11:
What hardware do you have?

I am guessing that you bought a hardware suited for GRBL 0.9 and are now trying to use GRBL 1.1 which has a new pin mapping?

See the documentation for GRBL 1.1 here:
https://github.com/gnea/grbl/wiki/Connecting-Grbl

And GRBL 0.9 here:
https://github.com/grbl/grbl/wiki/Connecting-Grbl

Jandre3d, 2024-04-11:
I am using a arduino uno wirh a cnc shield v3

breiler, 2024-04-11:
So you disabled this?
 ```
 // Enables variable spindle output voltage for different RPM values. On the Arduino Uno, the spindle
// enable pin will output 5V for maximum RPM with 256 intermediate levels and 0V when disabled.
// NOTE: IMPORTANT for Arduino Unos! When enabled, the Z-limit pin D11 and spindle enable pin D12 switch!
// The hardware PWM output on pin D11 is required for variable spindle output voltages.
#define VARIABLE_SPINDLE // Default enabled. Comment to disable.
 ```
Then the spindle on/off will be available on PIN 12. It seems impossible to find a proper wiring diagram of the shield, the best one I found was here: https://wiki.keyestudio.com/Ks0160_keyestudio_CNC_Shield_V3

You could contact the seller of the hardware for support.

Jandre3d, 2024-04-11:
Ohh that makes so much more sence now, yes I disabled that one, so I have to turn it back on and connect? my z limit switch to D12, and the + and - to D11 (z+)?

Jandre3d, 2024-04-11:
Oh here is the picture im referring to, forgot to include it in the previous comment 

https://images.app.goo.gl/ZukZqoRGtMcdmw2m8