grbl/grbl #1889: Problem load firmware Arduino IDE

ComputerGarageITA, 2024-07-04:
```
C:\Users\ComputerGarage\Documents\Arduino\libraries\grbl/nuts_bolts.h:51:0: warning: "max" redefined
 #define max(a,b) (((a) > (b)) ? (a) : (b))
 
In file included from sketch\sketch_jul04a.ino.cpp:1:0:
C:\Users\ComputerGarage\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/Arduino.h:93:0: note: this is the location of the previous definition
 #define max(a,b) ((a)>(b)?(a):(b))
 
In file included from C:\Users\ComputerGarage\Documents\Arduino\libraries\grbl/grbl.h:43:0,
                 from C:\Users\ComputerGarage\Desktop\sketch_jul04a\sketch_jul04a.ino:1:
C:\Users\ComputerGarage\Documents\Arduino\libraries\grbl/nuts_bolts.h:52:0: warning: "min" redefined
 #define min(a,b) (((a) < (b)) ? (a) : (b))
 
In file included from sketch\sketch_jul04a.ino.cpp:1:0:
C:\Users\ComputerGarage\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/Arduino.h:92:0: note: this is the location of the previous definition
 #define min(a,b) ((a)<(b)?(a):(b))
 
In file included from C:\Users\ComputerGarage\Documents\Arduino\libraries\grbl/grbl.h:43:0,
                 from C:\Users\ComputerGarage\Desktop\sketch_jul04a\sketch_jul04a.ino:1:
C:\Users\ComputerGarage\Documents\Arduino\libraries\grbl/nuts_bolts.h:55:0: warning: "bit" redefined
 #define bit(n) (1 << n)
 
In file included from sketch\sketch_jul04a.ino.cpp:1:0:
C:\Users\ComputerGarage\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/Arduino.h:124:0: note: this is the location of the previous definition
 #define bit(b) (1UL << (b))
 
C:\Users\ComputerGarage\Documents\Arduino\libraries\grbl\stepper.c: In function 'st_wake_up':
C:\Users\ComputerGarage\Documents\Arduino\libraries\grbl\stepper.c:192:8: error: 'stepper_t {aka struct <anonymous>}' has no member named 'dir_outbits'; did you mean 'step_outbits'?
     st.dir_outbits = dir_port_invert_mask;
        ^~~~~~~~~~~
        step_outbits
exit status 1

```

Mohosin353, 2024-08-27:
Mohosin

aafack, 2025-01-06:
I also encountered this problem. Have you solved it

Rob2029, 2025-11-25:
> . uecomment-3542569344
> I.  .  Me too. I also need help...
> I found a flashing tool. Give me your email address.

I

U