grbl/grbl #1894: How to disable sleep mode?

SebastianKusykIFJ, 2024-09-04:
Hello, I'm using GRBL for an XY scanner without any drill or laser. After about 15 minutes after sending the last command it enters something like a sleep mode, which makes it throw an error when the next command is sent. What's more important, GRBL looses it's coordinates in that mode/when exiting it. How can I prevent GRBL from entering sleep mode?

breiler, 2024-09-04:
GRBL is not entering sleep mode unless you tell it to, so this is a problem with either your hardware or OS.

Make sure your computer doesn't turn off USB peripherals power.

SebastianKusykIFJ, 2024-09-04:
GRBL is connected through Ethernet: there is a Waveshare Ethernet-UART converter.

breiler, 2024-09-04:
And how is your host computer connecting to it, is it connected over TP or WiFi? What software are you connecting with?

SebastianKusykIFJ, 2024-09-04:
Currently there is an Ethernet cable straight from PC to Waveshare.

breiler, 2024-09-04:
And you have tried ruling out the network part by connecting through USB serial?