grbl/grbl #1895: GRBL UGS Homing problems

pmozz331, 2024-09-12:
I saw a similar forum from 10ish years ago on this, so decided to start a new one to get some help with my issues.

I am using an arduino UNO, universal gcode sender and GRBL to drive a two axis machine. I have edited the config.h file to remove the Z axis from homing sequence, so it now reads;
#define HOMING_CYCLE_0 (1<<X_AXIS)
#define HOMING_CYCLE_1 (1<<Y_AXIS)

I am having some issues with my limit switches and homing cycle, however outside of this things are working well. I am using 1/8 microsteps, and my machine is calibrated to the correct steps/mm and I have adjusted speeds and accelerations to suitable levels. When I jog or provide commands like "G0 X100" in G91 it all works well, except for my limit switches and homing.

I thought that I had wired my limit switches as NO, however when I have $5 = 0, the machine will move about 5mm before giving the error:

An error was detected while sending '$H': ALARM: Homing fail. Streaming has been paused.
Error while processing response <ok>: An unexpected command was completed by the controller.

However when I have $5 = 1 or $5 = 2 the machine will move at the homing rate in the correct homing direction, but when it hits the limit switches will not stop.

I think at this point it's fairly obvious my limits are not right, but I am not sure of the issue and have attached images of the setup. I think this is the case because 1. when normally streaming, if I hold down one of the limits the machine won't shut off and 2. when in the setup wizard and I hold down the limits it won't register anything. How could I test my limit switches? And what is the reason for the strange behaviour with no motion for $5 = 0 but movement for $5 = 1?
On another forum, I saw that maybe this is an issue with the controller because when finished for the day I haven't shut down and disconnected in properly, I would appreciate your thoughts :)

Thanks for your time and advice, apologies for the long read I just wanted to make sure I included all info.

[image]
[image]

pmozz331, 2024-09-12:
Additionally:
When I test the switches with a multimeter they provide a circuit when open and no circuit when closed, all the way up to the board. When I connect them to the pins on the board, the multimeter beeps for a circuit constantly whether the switch is open or closed, I guess because the board pins must be connected.

pmozz331, 2024-09-12:
> It's been a very long time since I've commissioned a grbl system. A couple thoughts. 1. Limit Switches should be wired in a NC configuration, this would prevent the machine from moving if the switch wires lost contact. Standard in the CNC world. I'm not sure if this is your problem, but it couldn't hurt to test one. 2. You tested your switches with a multimeter, but can you confirm that the digital input signals are changing on the controller? 3. Your last comment is interesting, "the multimeter beeps for a circuit constantly whether the switch is open or closed, I guess because the board pins must be connected". This is something to inspect closer, if you put your mutli-meter on the pins that your switch terminates to (without the switch), do you have continuity? You should definitely not have it with your NO switch open and connected to board... (check XY and Z limit pins for continuity)
> […](#)
> -Scott

Apologies, I just googled the limit switches I have and they are wired in the NC configuration. Thanks very much for the reply, tomorrow I will have a look at the pins directly with the multimeter.

Also, no the digital input does not change in GRBL or UGS, I haven't looked in the arduino IDE so not sure about there. I have triple checked that I have hard limits enabled, and can't see any issues with the connections between the limits and the boards so am not sure why the signal isn't getting through. Thanks again for your help, will update with more info once I test.

pmozz331, 2024-09-12:
Ok so...
I am using a V3 CNC shield, and each of the limit switches connects to the limit pins (white row in the previous photo) and a ground (black row in the previous photo). On both sides of the board, I can't see any shorts with solder or anything like that. However, the X and Y axis limit pins are connecting to ground somehow, these are the limit pins I have been using. The Z-axis pins don't connect to the ground, I haven't been using these.
Have I somehow unknowingly shorted these pins?
The only other difference is that there are stepper motor drivers connected to the X and Y axis, but not to the Z, although I am not sure how this would change things...

scottrcarlson, 2024-09-13:
At least you’ve narrowed it down to hardware! I can’t comment on the board
as I’ve never had that one.  This certainly explains your issue though.