grbl/grbl #1903: Homing and Limit Switch Issue with GRBL 0.9j on Custom 2-Axis CNC Setup (X & Y Only)

DhaaraniMohanasundaram, 2025-02-07:
Hello,

I'm facing an issue with homing and limit switches on my custom 2-axis CNC setup (X and Y only). I am using GRBL 0.9j and Universal G-code Sender (UGS). Despite following standard setup and wiring instructions, I'm encountering problems with homing, limit switch detection, and hard limit alarms. Below are the details:

Setup Overview:
Firmware: GRBL 0.9j
Controller: Arduino-based CNC Shield
Machine Type: Custom 2-axis CNC (X and Y only, no Z axis)
Limit Switches: Normally Open (NO) switches (connected using C and NO pins, not NC).
Software: Universal G-code Sender (UGS)

Steps Taken:
Limit Switch Wiring: I have connected the limit switches to the C (Common) and NO (Normally Open) pins on the CNC shield. The NC pin is not connected.
Limit Switch Test (Multimeter): I verified the switches using a multimeter in continuity mode, confirming they work correctly.
GRBL Settings: I’ve configured the necessary settings in GRBL using $ commands in UGS, and the values appear to be correct (e.g., homing cycle enabled, limit switch pins configured).
Config.h Modifications: I made some changes to config.h in GRBL, but I couldn't find the specific LIMIT_PIN_MASK or equivalent settings for limit switches. I’ve checked other settings in config.h related to limit switch behavior, including homing direction and limit switch inversion.
Issue with Homing: When I press the Home button in UGS, the machine attempts to home but fails and gives an alarm:
Error message: An unexpected error was detected while sending '$H': ALARM: Homing fail. Streaming has been paused.
If I press any limit switch, it simply triggers an alarm without providing information on which switch was pressed.
Hard Limit Alarm:I also get a Hard limit alarm when pressing any of the limit switches:
Error message: An unexpected error was detected: ALARM: Hard limit
[Reset to continue]
Configuration Check: I checked $21 and $22 to ensure hard limits and homing cycles are enabled:
$21=1 (Hard limits enabled)
$22=1 (Homing cycle enabled)
Pin Assignments:
The limit switch pins are connected to X_LIMIT_PIN and Y_LIMIT_PIN on the CNC shield, but the exact pin configurations weren’t found in config.h. I believe these need to be correctly mapped to the corresponding physical pins on the shield.

Problem:
Limit switches are not being detected properly when pressed. Instead, I get a generic Hard limit alarm in GRBL.
The Homing function is not working as expected and results in a Homing fail alarm.
UGS doesn't show which limit switch was pressed when using the ? command.
I need to set a specific home position (X0, Y13) but don't know how to configure UGS to home to this specific position every time.

What I've Tried:
Rechecked wiring for the limit switches and ensured they are correctly connected.
Checked GRBL settings using $ commands to verify that the machine is set up correctly.
Tested with hard limits disabled to see if the limit switches were being detected (no change).
Checked config.h for correct limit switch pin assignments and inverter settings.
Tried homing the machine with the default configuration settings.

What I'm Looking For:
Correct GRBL settings and suggestions for limit switch pin assignments in GRBL 0.9j for X and Y axes only.
Solution to configure UGS so that homing will go to the correct position (X0, Y13).
Help in resolving the limit switch detection and Homing fail alarms.
Limit switch feedback in UGS (i.e., identifying which limit switch was triggered).

Additional Information:
I am using GRBL 0.9j, and I’ve made changes in config.h to suit my custom CNC setup with only X and Y axes.
The issue persists even after trying to reset GRBL settings.
Any help or guidance would be greatly appreciated!

Environment:
GRBL Version: 0.9j
CNC Controller: Arduino-based CNC shield
Limit Switch Type: Lever type (Normally Open (NO))
Software: Universal G-code Sender (UGS)

DhaaraniMohanasundaram, 2025-02-07:
I need assistance as soon as possible to resolve this issue and move forward with my project.

Please help ASAP!!!

DhaaraniMohanasundaram, 2025-02-07:
FOR REFERENCE FROM CONSOLE: 

>>> $H
An error was detected while sending '$H': ALARM: Homing fail. Streaming has been paused.
Error while processing response <ok>: An unexpected command was completed by the controller.
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]
Grbl 0.9j ['$' for help]
['$H'|'$X' to unlock]
>>> $X
[Caution: Unlocked]
ok
>>> $$
$0 = 10    (step pulse, usec)
$1 = 25    (step idle delay, msec)
$2 = 0    (step port invert mask:00000000)
$3 = 0    (dir port invert mask:00000000)
$4 = 0    (step enable invert, bool)
$5 = 0    (limit pins invert, bool)
$6 = 0    (probe pin invert, bool)
$10 = 3    (status report mask:00000011)
$11 = 0.010    (junction deviation, mm)
$12 = 0.002    (arc tolerance, mm)
$13 = 0    (report inches, bool)
$20 = 0    (soft limits, bool)
$21 = 1    (hard limits, bool)
$22 = 1    (homing cycle, bool)
$23 = 3    (homing dir invert mask:00000011)
$24 = 25.000    (homing feed, mm/min)
$25 = 500.000    (homing seek, mm/min)
$26 = 250    (homing debounce, msec)
$27 = 3.000    (homing pull-off, mm)
$100 = 250.000    (x, step/mm)
$101 = 250.000    (y, step/mm)
$102 = 250.000    (z, step/mm)
$110 = 500.000    (x max rate, mm/min)
$111 = 500.000    (y max rate, mm/min)
$112 = 500.000    (z max rate, mm/min)
$120 = 10.000    (x accel, mm/sec^2)
$121 = 10.000    (y accel, mm/sec^2)
$122 = 10.000    (z accel, mm/sec^2)
$130 = 6.900    (x max travel, mm)
$131 = 20.000    (y max travel, mm)
$132 = 1.000    (z max travel, mm)

breiler, 2025-02-07:
For this to work properly with UGS you will need to upgrade to version 1.1 of GRBL:
https://github.com/gnea/grbl/

The limit switches can only be detected from the status report in GRBL 0.9 if configuration activated for it:
https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#10---status-report-maskbinary

DhaaraniMohanasundaram, 2025-02-07:
There are various  1.1 version like e,f,h which will be more compatible for my project.

In grbl0.9,  i have tried to activate the limit switch configuration like $10 = 16 but still it only gave hard limit alarm like this:
An unexpected error was detected: ALARM: Hard limit
[Reset to continue]

any other ways to solve my problems using grbl0.9.

closed 2025-02-10 by DhaaraniMohanasundaram